prusa3d/PrusaSlicer #1003: IDEA: Have a  (or several) safe point(s) that nozzle moves to

boelle, 2018-06-24:
just an idea when filament runout happens

move to a safe point inside the object before lifting the nozzle

my thought behind this is to try and make it less visible that a runout has happended

maximlevitsky, 2018-06-30:
That would be awesome. In fact I would do the same for the print pausing, as it tends to pause on the perimeter. I would define per layer a safe point inside the object, as you say, and stop there.
Also the extruder  tends to leak, so again its best if it leaks to the object inside.

stephanpark, 2018-06-30:
@maximlevitsky
I haven't dug into the code but I'm sure run-out is a "Pause".  Pause does so soon as possible.  Since the print buffer is usually a continuous extrusion or avoidance travel move, it really shouldn't be broken up.  More specifically, it shouldn't be added to with an extraneous move in the case of a manual Pause which one assumes is desired immediately.  That shouldn't change.  So, "Pause Now" or "Pause Safe" might be a development path.

@boelle
But, as filament run-out is an automated user intervention request as opposed to user intervention event (not sure official terms), the firmware/G-code can't assume user will be immediately available thus automated Hold-For-User events should move to a safe zone of some kind.

At the moment (1.4), Pause seems to:
Retract (defined in G-code or overriden to 0.8 if none)
Lift extruder ~5mm.
Shut-off extruder heater.
Keep heatbed on at "Other" (not first layer).
Travel X outside the print min/max by ~10.
Travel Y to full aka present-bed.
Notes recovery line in G-code in memory.

Revision of Pause:
Multi-stage heating.  10 seconds full heat, 30 seconds bed only, 3 minutes all off.  User intervene cancels heat-standby routine to print heat.
Retraction and Lift head 25mm (mind max).
PAUSE_SAFE goes to 2nd infill line start point or if none default pause XYZ.
PAUSE_NOW stops at end-of-line, moves to default pause XYZ.

closed 2019-03-19 by boelle