prusa3d/PrusaSlicer #1008: HTTP 200 Error when sending gcode to OctoPrint

stelgenhof, 2018-06-27:
### Version
1.40.1-beta

### Operating system type + version
MacOS 10.13.5

### Behavior
When sending the gcode to OctoPrint results in an HTTP 200 error. (See attached screenshot).

### Steps to reproduce
1. Load an STL file (any will do)
2. Press the "Send to Printer" button 
3. Once the (new) popup appears, print "Ok"
4. HTTP 200 error appears (Body size exceed limits).

[image]

In most cases, the first attempt generates this error. Trying a few times again to send to the printer, the action is successful. Also had a other STL that worked at the first attempt though.

Note: Tested this on my Linux machine (Arch Linux) and works fine on that.

bubnikv, 2018-06-27:
@stelgenhof 

Does this happen with the Slic3r-1.40.0 final release as well, or is it a new bug?

@FidelCapo reported, that it happens quite often to him, if he checks the "print now" checkbox when sending to OctoPrint, but it goes usually through when the checkbox is not checked. Is it your case as well?

Thanks for the testing.

vojtechkral, 2018-06-27:
Thanks for the report, I think I know what the problem is, a fix is now being tested.

stelgenhof, 2018-06-27:
@bubnikv I believe it is not happening with the 1.40.0 final release. The issue happens either way: if the checkbox is ticked and not ticked.

Just to make sure: I have only noticed the issue on Mac OS. On my Linux machine it works fine. Can't say for Windows as I don't use it :)

PR #993 (2018-06-27): Bugfixes for 1.40.1 
- Fix ConfigWizard activating testing vendors
- Fix PresetUpdater displaying profile update _and_ application update on top of each other
- In ConfigWizard, append a `(default)` label to the first variant of each printer model to make some of the users less confused
- Fix #1008
closed 2018-06-27 by bubnikv