prusa3d/PrusaSlicer #1009: Cut function modifies objects

rfuehrer, 2018-06-28:
### Version
1.40.1-beta

### Operating system type + version
MacOS 10.13.5

### Behavior
1. Load attached STL two times (2x)
2. Select one of these thing (1x)
3. Cut the selected thing
4. Select Z=5mm (thing is 10mm in original)
5. keep both items after cut
6. slice your work and change to preview view
7. side of one item is seperated from body - **the upper part is fine, the lower part is broken** 
8. refer to second imported STL, this one is fine

Optional:
9. now cut the second STL (this one kept in original state)
10. select Z=4mm (thing is 10mm in original)
11. keep both items after cut
12. slice your work and change to preview view
13. in both items (upper/lower) one side is seperated from body

The seperated part (side) of the body is an added component to the main body in Fusion360. The seperated part is an 2mm addition to correct deviations.

Bugs:
- separation of parts from body
- uneven handling cutting of parts (5/5mm vs. 4/6mm)

_Is this a new feature request?_
no

#### STL/Config (.ZIP) where problem occurs
[image]

[terrace_ceiling_hook_extended_2mm.stl.zip](https://github.com/prusa3d/Slic3r/files/2144321/terrace_ceiling_hook_extended_2mm.stl.zip)

bubnikv, 2018-07-02:
The cutting will work, if you provide Slic3r with a correct geometry. Please try to fix your geometry with https://tools3d.azurewebsites.net/ . On Windows 10, we have integrated this service directly into Slic3r.

bubnikv, 2018-07-21:
This is a similar issue to https://github.com/prusa3d/Slic3r/issues/105 , the cut function does not work well on invalid or degenerate geometry.

rtyr, 2020-03-29:
I tested the cut function with this particular model and it works fine after fixing the model by Netfabb service. Closing.

closed 2020-03-29 by rtyr