prusa3d/PrusaSlicer #1011: Enhancement feature request - Perimeter count varying by layer

asieving, 2018-06-28:
I'd like to see an option so that the perimeter count could vary on a layer-by-layer basis.
It seems that if the perimeter vs infill boundary were to change on each layer, that there would be better attachment between the infill and the perimeter.  I see this being useful where you have an interior hole for a screw, you'd like the perimeter of the hole to be strongly anchored into the rest of the part.

E.G.
If you had a min_perimeter of 2 and a max_perimeter of 4, then the perimeter count might go like this:
4,3,2,3,4,3,2,3,4... for successive layers.  Having the infill and the perimeter varying should improve the bond between the two regions.

x0rtrunks, 2018-07-15:
Cura has a similar option to alternate adding an extra perimeter. It works quite well.

wavexx, 2019-10-15:
Cura calls this "alternate extra wall", which is nicely documented here: https://ultimaker.com/en/resources/52343-shell

jovannix, 2023-03-02:
you closed the new feature request. Do you already work on it or is it not possible to implement this feature?

darealfive, 2024-12-27:
This would be a nice Feature. 
Would this be added?

darealfive, 2026-09-02:
> In an attempt to clean up the GitHub issues tracker any legacy issue with no new user comments after this message will be automatically closed after 3 weeks.

Here I am, adding a user comment to prevent this ticket from being closed...
To change the perimeter count on a specific layer(range) would still be helpful.

otaolafranc, 2026-09-02:
> Here I am, adding a user comment to prevent this ticket from being closed... To change the perimeter count on a specific layer(range) would still be helpful.

one should not comment to keep it open if it is NOT an issue and a feature request but close it and open a feature request. that i already created for this issue https://github.com/prusa3d/PrusaSlicer/discussions/15660

darealfive, 2026-09-02:
@otaolafranc You could just as easily have said nothing rather than making me look as though I’d done something wrong, given that you were the one who’d just created the feature ticket.
How was I supposed to know that we weren’t supposed to reply here (anymore)? The ticket has been open since 2018 and there’s been no new information apart from this AI message.
I’ll carry on doing it this way in future too.

otaolafranc, 2026-09-02:
> [@otaolafranc](https://github.com/otaolafranc) You could just as easily have said nothing rather than making me look as though I’d done something wrong, given that you were the one who’d just [...]

You could had read the prusa slicer github policy on the original comment of the bot where they mention this. Commenting as you did will make that the issue stays. Which is undesired due to now the issues only going to be bug reports and everything else going to discussion function of the git. So how you were supposed to know by reading the bot comment and clicking in the link it gave you. The bot comments about if it is a bug comment if not it will close after by missing comment. Now I'd the original user of this issue is not here anymore and does not close it manually, the folks at prusa will net to go though and close it giving extra work to them.  Do whatever you want if you don't respect the time of others its your problem. They expressed clearly "hey please do this way" and you answer "I will keep doing my way" great folk.

asieving, 2026-09-03:
I am the original author of this issue, and since the prusa github has been reorganized, and since this issue is now covered by https://github.com/prusa3d/PrusaSlicer/discussions/15660, I am going to mark it as closed.

closed 2026-09-03 by asieving