prusa3d/PrusaSlicer #1012: contact z distance

BrandonMaynard545, 2018-06-29:
### Version
1.40.0+win64

### Operating system type + version
windows10+17134.112

### Behavior
Slic3r is adding extra space between the print and the raft. Even with the contact z distance set to 0

turn on raft it doesn't appear to matter how many layers

  * _If this is a command-line slicing issue, include the options used_
(umm this is a included option not command-line)

when contact z distance is set to 0 there should be no gap at all between the object and raft

Slic3er PE is adding .4 mm layer height gap even with contact z distance set to 0 

[image]

[image]

no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
cant export in slic3er the new update reverts all changes when exporting 

++++++++I would love it if you could possible email me at Sup454545@yahoo.com if you will be looking in to this issue it would mean a lot to know that it is being looked at++++++++

vojtechkral, 2018-06-29:
This seems to be a dupe of #926 I think ...

rtyr, 2020-03-29:
There shouldn't be a gap with Contact Z distance set to zero in the current version of the PrusaSlicer. Logic behind contact z distance is described here https://github.com/prusa3d/PrusaSlicer/issues/102,  https://github.com/prusa3d/PrusaSlicer/issues/926. Closing.

closed 2020-03-29 by rtyr