prusa3d/PrusaSlicer #1014: Browse for Octoprint crash

ctgreybeard, 2018-06-29:
### Version
1.40.1-rc+

### Operating system type + version
MacOS 10.13.5

### Behavior
On Printer Settings tab

Browse for Octoprint printer. Select one, click OK, app crashes. 

#### STL/Config (.ZIP) where problem occurs

No STL loaded, configuation is stock, not changed, recently updated, clean new install.
[wcw_1.40.1-rc.octoprint-crash.zip](https://github.com/prusa3d/Slic3r/files/2149474/wcw_1.40.1-rc.octoprint-crash.zip)

ctgreybeard, 2018-06-29:
This crash also happens when an IP is manually entered into the IP field for Octoprint. 

For me this is a blocker.

ctgreybeard, 2018-06-29:
I have attach one of the crash reports. They all appear to be in the perl GUI routines.

[perl5.26.1_2018-06-29-095630_ruusan.crash.zip](https://github.com/prusa3d/Slic3r/files/2149686/perl5.26.1_2018-06-29-095630_ruusan.crash.zip)

bubnikv, 2018-06-29:
Thanks, I am at home due to family issues, so we will fix it on Monday.
Thanks for vigilance. We have removed the OSX build from the release page
with the reference to the last two Slic3r issues. It seems we were overly
optimistic and we did not test the last fixes enough.

bubnikv, 2018-07-03:
Fixed with 1.40.1-rc2. Closing.

closed 2018-07-03 by bubnikv