prusa3d/PrusaSlicer #1015: Feature Request: Allow option to use filement color in place of extruder preview color in GCode preview for multi material prints

jimmy00784, 2018-06-29:
### Version
1.40.0

### Operating system type + version
Arch Linux kernel 4.17.2

### Behavior
New feature request:
Currently, in GCode Preview for multi material prints, Slic3r uses colors assigned in Extruder settings.
It would be more convenient to allow changing that to the filament color also so that the preview is a closer representation of what the final print might look like.

supermerill, 2018-06-29:
I don't understand. I use the same pla filament setting for each extruder.
Changing the color is just1-clic on the color icon in the main tab.

what is the current problem?

jimmy00784, 2018-06-30:
See the attached images. 
In the first, both extruders have same preview color of white. I am using Black PLA on E1 and Red PLA on E2. The preview generated shows everything in white - the preview colors on the Extruders 
[image]

In the second screenshot, E1 has Red preview color and E2 has Blue. I am still using Black PLA on E1 and Red PLA on E2. The preview generated shows E1 in Red and E2 in Black - as expected - the preview colors on the Extruders.
[image]

I am requesting that there also be a View where instead of the using Extruder preview colors, PLA colors are used so that the preview looks similar to what the print would look like.

I hope the explanation helps with the clarification.

supermerill, 2018-06-30:
You are making a different for each color because they need different cooling/temp?

The "filament color" as defined in the filament setting is only used as visual help. Maybe it should be removed to avoid confusion.

You wanted a checked box in the filament settings that, if checked, use the interface color as default filament color?

note: You have to click on the color icon (next to 'Filament:') to change the color for rendering. (sorry, but I don't know if you know it)

jimmy00784, 2018-06-30:
@supermerill The "filament color" and "extruder preview color" are both visual helps however in the preview render there is only the option to preview with extruder preview colors by selecting the "Tool" view.

> You are making a different settings for each color because they need different cooling/temp?

That could be a reason. I do a couple different brands for filaments. However, since I can specify a color for the filament in settings, I would like to see it being used when rendering the model after  it is sliced. I selected black filament for first tool and red filament for second tool. However, in the preview the calibration cube is rendered in red and blue based on the extruder preview color. 

> You want a checked box in the filament settings that, if checked, use the interface color as default filament color?

No. I can set the default filament color. I don't see anywhere it is used besides on the preview on the Filament dropdown.
What I would like to see is in the preview screen where currently you can see the "tool path" rendered with the extruder preview colors, also have the option to preview the tool path view with the filament color assigned on the filament.

>note: You have to click on the color icon (next to 'Filament:') to change the color for rendering. (sorry, but I don't know if you know it)

I am setting the color in the filament setting. It is visible in the on the filament selection

supermerill, 2018-06-30:
I'm sorry but I can't understand what is your problem. 
I hope someone else can help.

> No. I can set the default filament color. I don't see anywhere it is used besides on the preview on the Filament dropdown.

it's not a default color, consider it like a icon.

jimmy00784, 2018-06-30:
@supermerill There is no problem. It is a feature request. I just want the capability to use filament color instead of extruder preview color when generating the Gcode preview

tracernz, 2018-07-01:
In other words, a WYSIWYG view. I expected that's what the filament colour option was for when I first used Slic3r.

supermerill, 2018-07-02:
What we have right now (I now use icon for filament identification to avoid spelling confusion)

1) i load some pieces
[image]

1b) In the filament tab, i can change the icon of the filament type.
[image]

2) By clicking on the color icon, i can change the current color of the extruder
[image]

3) I have now sdelected my two color. Slic3r is showing me the two pieces in their respective colors (WYSIWYG ?)
[image]

4) Same for the preview thing
[image]

I proposed earlier to change the "icon" in filament setting to a "default color" but you said it's not what you want.

Could you show what you want in more details?

jimmy00784, 2018-07-02:
@tracernz @supermerill  The filament color does work if you haven't set "Preview Color" for Extruder settings. Then there is no way to use Filament color in Preview unless you delete the value from the config manually.
This below code is responsible for that behavior. If you notice extruder color has to be undefined or not a valid 6 digit hex entry.

`Slic3r/lib/Slic3r/GUI/Plater/3DPreview.pm:355`
```    # Collect colors per extruder.
    my @colors = ();
    if (! $self->gcode_preview_data->empty() || $self->gcode_preview_data->type == $self->{tool_idx}) {
        my @extruder_colors = @{$self->{config}->extruder_colour};
        my @filament_colors = @{$self->{config}->filament_colour};
        for (my $i = 0; $i <= $#extruder_colors; $i += 1) {
            my $color = $extruder_colors[$i];
            $color = $filament_colors[$i] if (! defined($color) || $color !~ m/^#[[:xdigit:]]{6}/);
            $color = '#FFFFFF' if (! defined($color) || $color !~ m/^#[[:xdigit:]]{6}/);
            push @colors, $color;
        }
    }
```
A much cleaner and user friendly approach would be to use a checkbox that set a flag to determine which colors to use.

jimmy00784, 2018-07-02:
This is what I am requesting. I modified the 3DPreview.pm file to get the desired effect. 
The setting isn't saved in the configuration file, but it is helpful even as a volatile setting

[image]

supermerill, 2018-07-02:
Ok, i understand
Maybe an other entry in the "view" combobox?

jimmy00784, 2018-07-02:
@supermerill Yes, that's what I had in mind when I posted the original feature request. 
Instead of the checkbox, a selection in the view combo box called "Filament" or some phrase more meaningful.
I might be mistaken, but I think with the dropdown box, there might be C++ code that might have to be modified. But with the checkbox that way I have it working now, it's all in the perl script.

supermerill, 2018-07-02:
Done

you can cherry-pick this commit: 619b4ef7256cc0129e6a4b468213d5192710bb8f before compiling your slic3rPE.

[image]

If @bubnikv is ok, i can make a pull request.

jimmy00784, 2018-07-02:
@supermerill Perfect. I think that'll do. Thanks a bunch.

bubnikv, 2018-07-02:
Actually @jimmy00784  is right about my original intention :-) I hoped to have a UI to set the extruder color to undefined, which would lead Slic3r to use the filament color instead. But this was never implemented, as I did not have time for that and nobody asked for it. Now as we have more people on board, we should do something about it.

I am not sure whether the solution proposed by @supermerill  is the one we want. I will have to sleep on it and discuss it with the team what is their opinion.

wsmagee, 2018-11-24:
I create a filament profile for each color of my filaments.  So having this feature implement in the future would be a benefit for me as well.  That way when I use the MMU2 printer profile, I can pick the filaments and would not also have to click on the color and set the extruder color to match so my visualization is color correct as well.  If I don't do this manually now, I can get my colors mixed up pretty easily when matching the parts of the object to the correct extruder.

jimmy00784, 2018-11-24:
@wsmagee as long as you don't set default extruder preview color, filament color is used in the gcode preview. Give that a try.

closed 2018-11-24 by jimmy00784

wsmagee, 2018-11-24:
@jimmy00784 .   Ok, thanks, I'm giving that a try but having issues.   I went to the Printer Settings, then to one of the extruders, and looking at the Preview section, Extruder Color field.  I think we are talking about this field.  If so,  it already has a color selected. (I am starting with the "Original Prusa i3 MK3 MMU2 printer profile).  I can not seem to find a way to delete the color, only change it.  Any thoughts on this?  
Thanks