prusa3d/PrusaSlicer #1020: Slic3r 1.40.1-rc fails to retain the filament list on

kursatu, 2018-07-02:
### Version
_Version of Slic3r used goes here_
1.40.1-rc
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
### Behavior
* _Describe the problem_
1- Open slic3r MMU
2- Select the filaments that are on your extruder.
3- Close sli3r
4- Open it again.
All 4 filaments are filled to be the same as the 1'st filament. 

* _Steps needed to reproduce the problem_
See the attached screenshot
[image]

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

kursatu, 2018-07-02:
Thanks for fixing I. I've noticed it in 1.40 but did not get around reporting it.
This might (and probably does) imply the following.
1- There is a testing hole on your end for MMU+UI because this could not have been possibly missed. 
2- There are few users of MMU.

bubnikv, 2018-07-03:
Fixed with 1.40.1-rc2

closed 2018-07-03 by bubnikv