prusa3d/PrusaSlicer #1021: Can we use a code formater?

supermerill, 2018-07-02:
I am a bit bothered
Every file has \t or 4 space or both and it's a mess.
Sometimes, braces are removed, sometimes they are here.
Sometimes there are the new foreach, sometimes the old for(begin;end;++)

Can we agree to a common coding style for the futures commits?
Maybe not enforcing it harshly but something to help people to contribute more easily, to keep a good base code.
Exemple: when using auto is okay, prefering using the foreach when possible, using tabs instead of spaces, using braces everywhere (even when there are only one statement after, because it's safer), inserting a space between each operator (=+*-/|&),  etc....
Because it's your code, it's up to you to decide what you want.

exemples: configWizard.cpp (some month old) and Serial.cpp (some month old also)

bubnikv, 2018-07-02:
> Can we agree to a common coding style for the futures commits?

That certainly makes sense.

> using tabs instead of spaces
> Every file has \t or 4 space or both and it's a mess.

Can't you set up your editor to interpret \t as 4 spaces? I don't think it is a good idea to fix this over the code base as it would introduce a huge amount of commits.

> using braces everywhere (even when there are only one statement after, because it's safer)

Why is it safer? It is longer and I don't like it.

> inserting a space between each operator (=+*-/|&), etc....

Some coding style will be helpful for sure.

The topic of coding style / coding guidelines is always controversial. I agree it would be great to have one, ideally adapting some widely used no-nonsense guideline. Do you have a proposal?

supermerill, 2018-07-02:
> Can't you set up your editor to interpret \t as 4 spaces? I don't think it is a good idea to fix this over the code base as it would introduce a huge amount of commits.

I did. I agree for the "fix over the codebase", it will create a big commit mess with "before" and "after", it's not nice. But at least, we should stop committing changes with the "not choosen one".

> Why is it safer? It is longer and I don't like it.
;p I know you don't like it (seeing your "cleaning" commits). I find it safer because it remove the burden to add/remove braces when we add /remove a statement in the bloc, and sometimes we forgot to add these braces, or a dev don't see the other statement and a bug happens. As you are (i think) the moderator of this repo, it's up to you to decide the rules, if you don't want it, it's ok for me, i'm not a fanatic.

>The topic of coding style / coding guidelines is always controversial. I agree it would be great to have one, ideally adapting some widely used no-nonsense guideline. Do you have a proposal?

I think we should stick mostly with what we have right now in the codebase (tab or 4 spaces, not 2 spaces). I can make a small proposal, but i think it have to please you (at prusa) before pleasing me.

stephanpark, 2018-07-02:
Maybe this is a philosophical argument.  Does poor etiquette ruin the chances of opportunistic participation of Open Source?  Or should the participant be responsible for adapting to the quirks of another contributor when the code improves the enterprise?

My opinion.  Write your own "readability" janitor program/script into your ingest/output workflow instead of trying to enforce arcane grammarian decrees that only serves to calm one's sense of aesthetic.  If these are paid individuals who's sloppiness detracts from legibility and thus smooth collaboration, then yes, wag fingers and at worst, terminate,  Open Source on the other hand is a "beggars can't be choosers" situation, absolutely no friction should be place between the desire to contribute and work being done to improve/debug the project.  If you simply can't contribute because tabs are spaces and the wrong space length at that, might I suggest you find out a little more about Regular Expressions.

bubnikv, 2018-07-02:
I believe we should introduce some coding guidelines. Ideally you would not recognize from the source code who wrote that particular section. But the coding style is always controversial and a matter of personal taste. For example, I prefer a compact code to fit as much as possible on the screen and commenting the code densely, while others prefer to use a lot of spacing and splitting the code into plenty of tiny functions with arguably descriptive names. Now imposing one's style onto others often leads to tension, so I would be careful about that.

If someone volunteers to write down the coding guidelines, we can certainly discuss it and try to codify something.

tracernz, 2018-07-02:
The Qt project are going through this same discussion at the moment [1]. It is just an contentious of an issue for them (as it has been in all projects I've seen). They are looking at adding a pre-commit hook that invokes clang-format to automatically reformat code to the project convention... but you need a clear and agreed convention before that can work. No matter what you do, automated formatting tools will always make some pieces of code less legible, e.g. messing up alignment between code and comments above if you're trying to illustrate something, but maybe it's worth it just to get that non-debatable consistency?

[1] Start of thread: http://lists.qt-project.org/pipermail/development/2018-June/032918.html

SijmenSchoon, 2018-07-10:
I am currently creating a clang_format which most closely matches the project's code style. I will report on how it's going when I make more progress.

SijmenSchoon, 2018-07-11:
Okay, after creating a clang_format which matches the code style, and trying to pick the most common option where the project is torn between two styles, here's a quick draft of the code style I propose, mostly based on the default clang_format coding style.

Note that this is just my proposal and of course really opiniated. Any feedback is really welcome. I tried to keep it as short as possible, only including the really necessary guidelines.

---

**Indents** are done using four spaces.

```cpp
while (condition) {
    do_thing();
}
```

***Note:** This was **very** controversial in the code base. I decided for spaces because of spaces being much easier to use for aligning things, and combining tabs and spaces is hard to do consistently.*

The code inside **namespaces** is not indented, to avoid unnecessary indentation inside. The closing brace has an inline comment with the name of the namespace.

```cpp
namespace Slic3r
{
class Foo
{
    // ...
}
} // namespace Slic3r
```

**Access modifiers** and **case labels** are not indented.

```cpp
class Foo
{
public:
    // Public members.
    
private:
    // Private members.
}
```

```cpp
switch (variable) {
FOO:
    break;

BAR:
    break;
}
```

There's a maximum of 100 **columns** on one line. This is a good compromise between readability and amount of linebreaks needed.

Every **statement** gets its own line.

```cpp
// Do NOT do this
do_a_thing(); do_another_thing();

// Do this instead
do_a_thing();
do_another_thing();
```

**Parameters** are put on a new line if they do not fit on a single line, without aligning to the opening bracket.

```cpp
void GLIndexedVertexArray::render(const std::pair<size_t, size_t> &tverts_range,
                                  const std::pair<size_t, size_t> &qverts_range) const
{
```

Similarly, **operands** of binary and ternary operations are aligned after newlines.

```cpp
int aaa = bbbbbbbbbbbbbbb +
          ccccccccccccccc;
```

**Consecutive assignments** and **declarations**, however, are never aligned. This is due to it being hard to draw the line what lines should be aligned, and which things on the line should be aligned.

```cpp
int aaaa = 12;
float b = 23;
std::string ccc = 23;
```

There are spaces around all **binary** and **ternary operators**.

**Pointers** and **references** are aligned to the right. This is because the pointer/reference token is parsed as if it's attached to the variable name (e.g. `int *foo, bar` makes `foo` a pointer, but not `bar`).

```cpp
int foo(int *param_a, int &param_b)
{
```

The **brace** breaking style of the Linux styling guidelines is used: attach all braces to the surrounding context, but break before braces on function, namespace and class definitions. Braces are left out wherever possible.

```cpp
namespace Slic3r
{
class FooBar
{
    void foo()
    {
        bar();
        
        while (true) {
            if (condition) {
                bar();
                baz();
            } else
                baz();
        }
    }
}
} // namespace Slic3r
```

It is okay to put very short functions, with only a single statement, on a single line:

```cpp
std::vector<float> get_extruders() const { return m_output_extruders; }
```

***Note:** This was controversial in the code base. I decided for the Linux style since it was just slightly more popular than the other options.*

**Includes** are sorted alphabetically:

```cpp
#include "AppConfig.hpp"
#include "../../libslic3r/Utils.hpp"

#include <stdexcept>
#include <utility>
#include <vector>

#include <boost/algorithm/string/predicate.hpp>
#include <boost/filesystem.hpp>
```

---

The `.clang_format` file for this:

```
---
IndentWidth: '4'
BreakBeforeBraces: 'Linux'
ColumnLimit: '100'
AllowAllParametersOfDeclarationOnNextLine: 'true'
AccessModifierOffset: '-4'
...
```

---

I think this will be a good start for cleaning up the project to have a much neater and especially more consistent code style, and for encouraging neat and consistent new code.

supermerill, 2018-07-11:
Good for me.

any opinion for /* */ comments vs  //  ?
In the sli3r repo, they allow only //

SijmenSchoon, 2018-07-11:
I don't really have an opinion on them. The most prevalent choice was `// comments`, so I suggest using those.

lordofhyphens, 2018-07-17:
@supermerill as a FYI for mainstream slic3r I prefer that developers only use `/* */`  style comments for commenting out code blocks (which should be done sparingly). 

Any comments in headers should be `///` (doxygen). In cpp files for local (in-method) commentary `//` is fine. 

@SijmenSchoon I'm stealing that clang_format and putting it into mainline slic3r for devs to use (and requiring it for c++). Thanks!

@supermerill if you see this problem in mainline Slic3r please open an issue for it.

lordofhyphens, 2018-07-17:
One legacy note (which is probably all over the prusa3d fork as well): @alexrj used 
````
void
foo()
{
}
````
in most of the code in libslic3r.

vojtechkral, 2018-07-18:
Is the 100 columns limit a hard one or a soft one? Occasionally it makes sense to me to make a longer line rather than splitting it if the code is dense already and needs more than regular eyeballing anyway...

bubnikv, 2018-07-26:
> I can see big ugly line, that i think are not making many sense:

That is an indicator, that Prusa Research cares for its employees, and we are working with a top notch hardware :-)

The line does not fit MY monitor, so is too long.

I just checked some of the lines I wrote, they are up to 180 characters long. It is difficult to work with such code on my laptop screen. The particular line you referenced is 232 characters long, but I have found one, which is 255 characters long. Clearly, @lukasmatena has a much better eye sight than me (he sits next to me and his monitor is the same size as mine).

```
std::vector<Region> &by_region_specific = island.by_region;
// some useful comment do write about the if like:
// if some regions are overriden, create a new vector without them inside
if(const_cast<LayerTools&>(layer_tools).wiping_extrusions().is_anything_overridden()) {
    by_region_specific = island.by_region_per_copy(copy_id, extruder_id, 0 != print_wipe_extrusions);
}
```
This does something different from the original code by @lukasmatena . Yours would overwrite the content of island.by_region, so it is not correct. IMHO @lukasmatena 's code is just fine, but with a worse eye sight, therefore a bigger font on my screen, I would break the line into two lines. But I think this is a detail, and correctness is much more important than following a precise coding guideline. You can always enable word wrapping in your editor.

> Also, can we ban the auto keyword, unless for special cases like for an iterator?

Would you enumerate the special cases? There are many cases, like 
```
auto v = new SomeClass()
```
where the use of auto is desirable, not just allowed.

I agree with @supermerill though, that in this particular case I would personally not use auto, as the return type is not obvious. We will discuss this topic internally.

supermerill, 2018-07-26:
> This does something different from the original code by @lukasmatena . Yours would overwrite the content of island.by_region, so it is not correct. IMHO @lukasmatena 's code is just fine, but with a [...]

It just updates the reference, which is used just after. and then out of scope. It add a single instruction, cached by the cost of the comparison, and maybe optimised by the compiler or the out-of-order pipeline.
Or maybe i need to do some c++ revisions :-p

supermerill, 2018-07-27:
ohhh, ok, i'm wrong.
I should have used a pointer.

SijmenSchoon, 2018-08-01:
I am really against a 160 character limit personally. Sure, it fits on wide monitors no problem, but:

1. side-by-side viewing becomes a pain
2. there (sadly) still are plenty of laptops with 1366x768 displays, on which you really don't want to fit more than 120 characters before it gets unreadably small.
3. lines are much more readable when kept short
4. if your line becomes as long as 160 characters, you really are stuffing way too much logic into a single line and should be split up into multiple lines by e.g. storing results into intermediary variables

tracernz, 2018-08-01:
> side-by-side viewing becomes a pain

Totally agree with everything you said. That is why a wide variety of projects have a limit of 80 columns (ISO standard terminal), or 100 columns. On my dev machine, I typically have 2x2 terminals (editors live in these) on the left 4/7ths and web browser on the right 3/7ths of the display. 100 columns is about the most I would like.

bubnikv, 2021-09-25:
We now have our coding guidelines

https://github.com/prusa3d/PrusaSlicer/wiki/Contribution-guidelines#coding-style

and @supermerill has his own superslicer where he could apply code formatter.

Closing.

closed 2021-09-25 by bubnikv