prusa3d/PrusaSlicer #1024: Missing 'undo' feature

cassianoleal, 2018-07-02:
### Version
All

### Operating system type + version
All

### Behavior
* _Describe the problem_
There is no way of going back after making a mistake or just quickly trying out something.

* _Steps needed to reproduce the problem_
  * Do stuff
  * Do other stuff
  * Realise "other stuff" was a mistake

* _Expected Results_
  * Press button or shortcut to undo the mistake

* _Actual Results_
  * Cry, take deep breaths, start over

_Is this a new feature request?_
Yes

vojtechkral, 2018-07-20:
Yes, this would be nice to have and we're looking into it, but I can't give a time estimate at this point...

cassianoleal, 2018-07-22:
This makes me a little sad.

I love the new features coming out of this team (kudos, there's some awesome stuff there!) but in my opinion Slic3r PE is already a good slicer feature-wise.

The UX is very much subpar though and should be looked into ASAP. Undo and keyboard shortcuts are crying out to be implemented.

photog0411, 2018-12-12:
Also wanting undo feature.

Official slic3r has it. Can you just merge it into this fork?

bubnikv, 2018-12-12:
> Official slic3r has it. Can you just merge it into this fork?

No, we cannot merge it, the two code bases diverged too much.

photog0411, 2018-12-14:
> > Official slic3r has it. Can you just merge it into this fork?
> 
> No, we cannot merge it, the two code bases diverged too much.

@supermerril

can you add this to your fork or is the answer the same

bubnikv, 2018-12-14:
We will implement undo/redo on our own independently from the upstream
slic3r.

NielschB, 2019-05-07:
This feature would be really helpful in my opinion. Not sure what the undo depth should be, but at least one operation would already help a lot!

guestisp, 2019-05-07:
I agree, especially for the cut operations or the delete button

liftbag, 2019-05-16:
Following and hoping for an even just simplified undo

sonictrain, 2019-05-16:
Essential feature! Pls implement it 😊

bubnikv, 2019-05-16:
There is always hope.
Die Hoffnung stirbt zuletzt.

bigsubgit, 2019-05-16:
Undo feature is very useful, please, let implement it.

photog0411, 2019-05-16:
Yeah implement dat stuff

cassianoleal, 2019-05-17:
I agree with @NielschB - even a single step undo would already be a lifesaver!

klingat, 2019-06-25:
NEED this feature! +1

lukasmatena, 2019-06-26:
We are already working on it.

steinerlein, 2019-07-04:
@lukasmatena can you try to estimate a possible release date? Thanks for working on this!

lukasmatena, 2019-07-11:
@steinerlein No. But we should manage to finish it for the next release.

bubnikv, 2019-08-08:
Implemented by the upcoming PrusaSlicer 2.1.0-alpha1
Closing

closed 2019-08-08 by bubnikv