prusa3d/PrusaSlicer #1025: [Linux] 1.40.1-RC screen issues with AppImage

ctgreybeard, 2018-07-03:
### Version
1.40.1-RC
### Operating system type + version
Linux Mint 19 (Ubuntu 18.04 base)

### Behavior

When starting the AppImage the initial screen gets corrupted, parts of the background get "slid" partially into place: [image] The app window itself is not displayed, only the plater subwindow.

Clicking the Menu button (bottom left on the screen) clears the corruption and displays the app window. [image] *Note, however, that the menu itself if obscured behind the Plater subwindow.*

Minimizing the app window leaves the Plater window on screen obscuring anything beneath it. It remains the top window regardless of anything else. [image]

When the app window is showing properly the Plater subwindow obscures any menu items that appear behind it. [image] This prevents, for instance, closing the application as the File->Quit menu item is hidden behind the Plater.

Moving off of the Plater tab restores function but things like Object Rotation cannot be performed because they only appear on the Plater tab but are obscured.

_Is this a new feature request?_ No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_ N/A, no STL is required to exhibit the issues.

ctgreybeard, 2018-07-03:
1.40.1-RC2 also exhibits the same issues.

RC2 generates the following to stderr/out:

```
Creating main frame...

(slic3r.pl:31645): Gtk-CRITICAL **: 13:32:56.685: IA__gtk_color_button_set_color: assertion 'color != NULL' failed
OpenGL Warning: Unknown program 3
OpenGL Warning: Unknown program 6
```

ctgreybeard, 2018-07-03:
I built Slic3r from master here and found a few missing dependencies:
```
libcurl-openssl-dev
libeigen3-dev
libglew-dev
﻿ExtUtils::Typemaps::Basic
﻿ExtUtils::XSpp::Cmd
```
but it built fine after that. Running it, however, gives an error:
```
╰─➤  perl slic3r.pl --gui
Can't load '/usr/lib/x86_64-linux-gnu/perl5/5.26/auto/OpenGL/OpenGL.so' for module OpenGL: /usr/lib/x86_64-linux-gnu/perl5/5.26/auto/OpenGL/OpenGL.so: undefined symbol: glGetFramebufferAttachmentParameteriv at /usr/lib/x86_64-linux-gnu/perl/5.26/DynaLoader.pm line 187.
 at /home/bill/Projects/Slic3rPE/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 134.
Compilation failed in require at /home/bill/Projects/Slic3rPE/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 134.
BEGIN failed--compilation aborted at /home/bill/Projects/Slic3rPE/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 134.
Compilation failed in require at /home/bill/Projects/Slic3rPE/Slic3r/lib/Slic3r/GUI.pm line 16.
BEGIN failed--compilation aborted at /home/bill/Projects/Slic3rPE/Slic3r/lib/Slic3r/GUI.pm line 16.
Compilation failed in require at (eval 77) line 1.
```

bubnikv, 2019-05-21:
Is the PrusaSlicer-2.0.0 AppImage working for you? Can we close this issue?

rtyr, 2020-07-29:
I will ask again before closing. Is the current PrusaSlicer 2.2.0 AppImage working fine for you?

ctgreybeard, 2020-07-31:
Sorry, I didn't see this. Yes, the appimage is working great

closed 2020-07-31 by ctgreybeard