prusa3d/PrusaSlicer #1028: Thin walls: improvement

supermerill, 2018-07-04:
Hi,
I just slain my personal dragon by improving the thin wall algorithm.

 * no more strange turn at end => fusion of voronoi end lines when they are near enough
 * anchoring inside the perimeters => extends the lines end inside the perimeter to join it. A gui parameter can be created to adjust it if needed (currently, it extends to meet the center).
 * conservation of too small segments of thin walls when they are crossing each other => modification of the fusion algorithm to select the most strait path instead of a random one
 * no more cutting of small walls path by the gcode writer (no_sort for each thin wall extrusion) to have nice long lines of extrusions instead of many smaller.

before:
[image]
[image]

after:
[image]
[image]

Do you want a pull request?

for the curious: f6854427eb8ab44c3beddda2dc5386c73cda08e3

lordofhyphens, 2018-07-13:
This has been merged into upstream Slic3r

bubnikv, 2018-07-17:
@lukasmatena Would you please review this code? Thanks.

supermerill, 2018-07-18:
If you want to make a code review, i should create a pull request as all the needed code isn't in this commit.

wavexx, 2019-02-22:
Would you make a proper PR for this? I think it would also be helpful also for prusa to keep track of other tweaks you did beyond the mentioned comment (if any).

supermerill, 2019-02-22:
I will update it when the one in the mainline branch will be merged.

wavexx, 2019-02-24:
There's so much development to love both in slic3r mainline and prusa's fork... :cry: 
I switch between the two depending on the part I need to print!

closed 2026-09-29 by prusaslicer-bot[bot]