prusa3d/PrusaSlicer #1029: Slicing in vase mode not honoured and failure to honour not detected

kortschak, 2018-07-04:
### Version
1.40.1-beta+linux64

### Operating system type + version
Ubuntu Linux 16.04.4

### Behavior
* Slic3r omits crucial surfaces of the model [here](http://www.thingiverse.com/thing:2277253) when being sliced at full size, but not when sliced above 135%. The model is designed to be printed in vase mode, but the surface omission results in a topology that cannot be printed in vase mode. No check is made to confirm that the vase mode setting is being respected by the slicer.
* Slice the model linked above at 100% with the spiral vase mode set.
* Expect to see single continuous path for extrusion of the walls (shown here by slicing at 150%).

[image]

* What I actually see is multiple retractions and travels as the internal moves are omitted;

[image]

This is not a feature request

#### STL/Config (.ZIP) where problem occurs
The only change to the config from stock is to set spiral vase. The STL is available from https://www.thingiverse.com/download:3618392. I can upload a zip of these if given a location that will accept large zip files - github rightly do not.

DrLex0, 2018-07-04:
Slic3r ignores gaps narrower than 0.1 mm, see #520. The gaps in this model are only 0.075 mm, which is why it works when scaling above 133.33%.
But you're right, there should be some kind of feedback if vase mode cannot be honored.

supermerill, 2018-07-04:
Slic3r remove gaps that are below a threshold, I think that's your problem. You have to create bigger gaps in the stl.

kortschak, 2018-07-04:
This is not my model and only the stl is published, so I'm not able to change the gap width. It seems though that this would be fixed if the gap threshold were configurable, or probably easier if vase mode forced a lower threshold. This latter option (cf the options mentioed in #520 of lowering it wholesale or making it configurable) seems better since the user is providing an assertion that the model is a single manifold surface. Add in a check that this is true and you have both this issue and #520 solved.

kortschak, 2018-07-04:
Even some documentation of the limit within Slic3r would be valuable.

DrLex0, 2018-07-04:
@kortschak: because I'm curious how well this will print, I have modified the model in Blender to make the gaps 100 micron wide. Perimeter extrusion width must be set to 0.4mm for vase mode to work. [ParacordVaseSpool-Slic3rCompatible.stl.zip](https://github.com/prusa3d/Slic3r/files/2164451/ParacordVaseSpool-Slic3rCompatible.stl.zip)

kortschak, 2018-07-04:
@DrLex0, thanks.

bubnikv, 2019-03-01:
By the way, I have introduced a new parameter "slice_closing_radius", which is set to 0.049 by default, leading to the same results as in 1.41.3. If you need to maintain the thin gaps, starting with Slic3r-1.42.0-alpha7 you will need to lower the new "slice_closing_radius" value.

Implemented with e3cba0a65a691b77be87f8440c91ec5f4916c3b8

This parameter and the return to the 1.41.3 behavior by default was made as we have discovered some models with cracks, which were handled correctly by 1.41.3, but which were full of spurious perimeters in 1.42.0-alpha5. We find it safer to revert to the 1.41.3 behavior while giving you the parameter to tinker with.

closed 2019-03-01 by bubnikv