prusa3d/PrusaSlicer #1031: Sli3r 1.40.1-rc2 does not retain purging volumes

kursatu, 2018-07-05:
### Version
_Version of Slic3r used goes here_
1.40.1-rc2 1.40.1-rc2 
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
1- Change the purging volumes
2- Exit and start sli3r
2- The purging volumes are lost and the original ones show up.

* _Steps  needed to reproduce the problem_
[image]

[image]

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-07-05:
The purging volumes are being stored into an AMF or 3MF file. Does it
fulfill your needs?

kursatu, 2018-07-05:
A good behavior can be modeled, for the .stl files.
Here is I think what could work.
- When I am operating in ".stl mode" meaning that I've opened an .stl file, 
the purge volumes changes will be retrieved from application settings and saved whenever I change it.
- In the AMF, 3MF mode, ie. when I opened one or save as one, the purge volumes associated with the
AMF,3MF file will be applicable.

Saving AMF and 3MF files is definitely needed.
But If I can't change the default settings it is bound to cause problems.
I'll have to change purging volumes every time when I open an .stl file, then at some point
I'll forget and have to restart a print. I am also printing fine with a lower purge volumes on a wide range of materials. So before I had to make a specialized change using 3MF,AMF files I'll use
a different setting, that is not changeable at this time.

bubnikv, 2018-07-17:
> Saving AMF and 3MF files is definitely needed.
> But If I can't change the default settings it is bound to cause problems.

I think you requested #1036, which will be implemented in 1.41.0-alpha. Will that satisfy your needs?

kursatu, 2018-07-17:
Yes, using the AMF file as a replacement for the capability of changing this application setting 
would probably work, if #1036 were to be fixed. 
I am still not using AMF because it is not really usable
because of #1036, so don't quite know if there are other similar defects.
Basically I'll need to use an AMF always just because I can't change this setting.
I'd still consider this issue a lower priority defect though. Not sure when it can be fixed, given the calculation of the effort fixing it vs. value/users effected.