prusa3d/PrusaSlicer #1033: Upgrade Issue: Missing Brim Setting Options for Single Object

carlvonkessler, 2018-07-05:
### Version
1.40.0+win64

### Operating system type + version
win10 home

### Behavior
* 1.40.x no longer provides item specific setting options such as brim options under the settings menu
* right click on object -> settings-> object settings.

This is a feature from previous versions that is no longer available. 

What if I need a brim on a single object and not the entire print?

#### STL/Config (.ZIP) where problem occurs

bubnikv, 2018-07-17:
Are you sure it is a regression issue for the Slic3r Prusa Edition? This looks like a new feature in the upstream Slic3r.

carlvonkessler, 2018-07-19:
Slic3r 1.29.x stable does lack a number of the settings choices for a single object including an brim.

bubnikv, 2018-08-02:
ok, for the Prusa Edition Slic3r it is a new feature, so I have re-classified it.

fwdixon, 2019-02-10:
I too would like to see this feature implemented.

jmckeown2, 2019-06-03:
Is this easier or harder to implement after the PrusaSlicer 2.0 restructuring?

carlvonkessler, 2019-06-03:
It’s much better with PrusaSlicer

thalesfsp, 2019-06-06:
@carlvonkessler Do you have this feature in the roadmap? Thanks!

gmccauley, 2019-07-03:
+1 for this feature

tdiggity, 2019-08-06:
+1 this would be great to have.

andrewluebke, 2019-08-10:
+1 this would be great for printing a multipart piece which only one item needs a brim.

Jebtrix, 2019-08-12:
+1 That would be one less workaround. For simply choosing individual pieces I currently have to do it this way. It works but saving print time of unused brim would be appreciated.

[image]

[image]

ManuGithubSteam, 2019-08-13:
I second this too

WarrenSchultz, 2019-11-10:
Tree supports sometimes could really use a brim, so this would be a big improvement for me.

audiorazor, 2019-12-18:
Would be awesome to implement this sooner than later.   Seems like it would be easy to add with a modifier but what do I know.

nellering, 2019-12-20:
+1 for this feature

ManIkWeet, 2019-12-22:
+1 I run into this too often

singultus, 2020-01-13:
Please make it compatible for multipart objects. While printing multicolour with mmu the single color brim leaves discoloration marks on the parts with a different co or.

codyolsen, 2020-02-03:
[image]

subsite, 2020-03-03:
+1 The new "Add settings" feature for separate objects is great, but it lacks the IMO most useful one.

jcwren, 2020-03-29:
+1 for this feature. Sure could use it right now.

Windscreen2011, 2020-04-09:
+1
One more upvote.  PS 2.2 seems to have per part settings for nearly everything, but this.

houmark, 2020-04-12:
Another request for per object brim.

koporcam, 2020-04-12:
> 
> 
> +1 That would be one less workaround. For simply choosing individual pieces I currently have to do it this way. It works but saving print time of unused brim would be appreciated.
> 
> [image]
> 
> [image]

> 
> 
> +1 That would be one less workaround. For simply choosing individual pieces I currently have to do it this way. It works but saving print time of unused brim would be appreciated.
> 
> [image]
> 
> [image]

Jebtrix, I must be missing what you are doing. Can you explain? Thanks!

ThePirat91, 2020-05-29:
+1 for this very useful addition

klsc18, 2020-06-21:
+1 need it many times I print multiple parts

narical, 2020-09-09:
> Edit: I figured out a method to force PrusaSlicer to sort of do what I want:
> 
>     1. Add PART shape, one layer thick (0.15mm here) and place it under your stubborn area
> 
>     2. Right click the part to add infill and layer/perimeter modifiers to it.
> 
>     3. Set 100% infill, no top/bottom layers, **NO PERIMETER**
> 
>     4. Slice
> 
>     5. Go back and adjust your modifiers how you like. Re-slice. You should get something like this:
Not working for me, sadly.
This issue is the only reason I intalled Cura slicer and trying to learn it ((

neophyl, 2020-09-09:
You could always use Super Slicer, which is a fork of Prusa Slicer which has had this feature for awhile as well as other brim tweaks.

ChiefPoints, 2020-09-11:
I run into this on a nearly daily basis, especially when trying to fit lots of smaller pieces on the build plate. Not having the ability to add a brim via a piece by piece basis seems like a very strange omission given the other modifiers we have access to in Prusa Slicer. Please add this in as soon as possible so we end up using one less workaround. Thanks!

wavexx, 2020-09-11:
> You could always use Super Slicer, which is a fork of Prusa Slicer which has had this feature for awhile as well as other brim tweaks.

Is there a specific commit that introduces this option? Maybe it's easy to backport.

neophyl, 2020-09-11:
Not sure wavexx. I do know that the whole brim functionality is different and imo much better. 
Firstly the brim moves with the part so even if you set a large elephants foot compensation it moves in with the first layer.  There is then a brim offset which moves it out from the first layer by a specified amount. This decouples it from elephants foot so no more ‘my brim isn’t attached issues’ while still allowing easy removal if that adjustment is needed. 

You then have all the other options, like brim on outside and inside options which is useful for things like tubes etc, there are brim ears which only adds brims at corners, you can nest a small part inside a larger part and still get a brim on it with one of the other options which you can not do in PS. 
I really like PS and all the work that has gone into it and use it a lot, but Super Slicer is like the tinkerers version, with loads more settings made available so if you like to experiment it’s just got more options.

mneumark, 2020-10-31:
Any word from Prusa on this? 2.3.0a2 seems to have added a lot of settings per object, but not brim....any chance this could be added in 2.3.0?

LuisHerrero92, 2020-11-02:
+1. Skirt option would be also appreciated in my case.

ThePirat91, 2020-11-02:
+1 for the brim options on 2.3.0

bubnikv, 2020-11-02:
@neophyl commented on Sep 9
> You could always use Super Slicer, which is a fork of Prusa Slicer which has had this feature for awhile as well as other brim tweaks.

Once you start experimenting with SS brim settings, you will soon find out that SS (and the upstream Slic3r) will happily extrude brim of one object over the brim of another object. Sure, sometimes it works correctly, but we already have enough of bug reports on supports of different objects overlapping. We cannot do that to our support guys to merge such a feature.

@hejllukas is currently working on a new implementation of brims per object / internal brims, that would not extrude one over the other. It is most likely though, that we will have to limit the brim width to a global value and just enable / disable external resp. internal brims per object.

narical, 2020-11-24:
> Once you start experimenting with SS brim settings, you will soon find out that SS (and the upstream Slic3r) will happily extrude brim of one object over the brim of another object. Sure, sometimes [...]

For now it's much better than not being able to do sequential printing at all. Now PS prints all brims for all objects first, then begins to print objects themselves - with bad adhesion to bed and brims. My prints fails so often that I almost forced to use another slicer. I tried SS and it works much better in that case. Can't describe how much happy I am not using Cura anymore!

zeno4ever, 2020-12-20:
+1 for the next PrusaSlicer version

stevenvo, 2021-02-09:
this should be added, +1

ThePirat91, 2021-02-09:
+1 for the next version

rogery555, 2021-02-13:
Insane how this is not an option

ffcruz85, 2021-02-17:
seems like a no brainer - there are modifier settings for way more advanced settings

astupidmoose, 2021-08-01:
Honestly I just assumed this would be a modifier.. was upset it wasn't lol.

lukasmatena, 2021-08-12:
Upcoming PrusaSlicer 2.4.0-alpha1 will allow to set brim for each object separately. Closing.

closed 2021-08-12 by lukasmatena

LuisHerrero92, 2021-08-12:
Skirt settings too? Thank you!

lukasmatena, 2021-08-12:
@LuisHerrero92 No, skirt settings will stay as they were. In fact, may I ask what is the benefit of having skirt separate for each object? I don't quite see it (unlike brim - that is clear).

LuisHerrero92, 2021-08-12:
Hi @lukasmatena
In my case -printing sequential objects with 1.2mm vulcano nozzle- an individual skirt serves to prime the nozzle (remove the oozing that appears from the travel from part to part and get a proper extrusion right from the start of the piece because the travel distance from the skirt to the piece is minimal) thus ensuring a successful first layer and part bottom.

I've tried calibrating ooze, retraction and extrusion, and I've also tried linear advance... All to no avail, the skirt before each part is what works for me and so far I'm doing it manually by chaining gcodes.

narical, 2021-09-22:
> Hi @lukasmatena
> In my case -printing sequential objects with 1.2mm vulcano nozzle- an individual skirt serves to prime the nozzle (remove the oozing that appears from the travel from part to part [...]

There's something with your print settings or wet filament (petg in particular), I have vulkan too and usually print without brims (with nozzles 0.4-1.2).There are issues sometimes but not a big deal.

dhlavaty, 2025-03-22:
> Upcoming PrusaSlicer 2.4.0-alpha1 will allow to set brim for each object separately. Closing.

I'm on 2.9.1 where is it, please ? Especially with multiple-beds this is a very important feature, and I could not find it anywhere.

neophyl, 2025-03-22:
Select the object, right click it and then add settings.  You must be in expert mode.

dhlavaty, 2025-03-22:
Wow, "expert mode" was the missing thing. Thank You. _I have to admit that I didn't expect the expert mode to change the drop-down menu items - I find it a bit unintuitive._