prusa3d/PrusaSlicer #1034: Slic3r 1.40.1 crashes when getting Info from Help -> System Info

mcstoufer, 2018-07-06:
### Version
_Version of Slic3r used goes here_
1.40.1
_Use `About->About Slic3r` for release versions_
Downloaded build from https://github.com/prusa3d/Slic3r/releases/download/version_1.40.1-rc2/Slic3rPE-1.40.1-rc2.full-201807031453.dmg
Placed App bundle on top of Slic3r version in the 2.1.7 bundle from https://prusa3d.com/downloads/drivers/prusa3d_mac_2_1_7.zip

### Operating system type + version
_What OS are you using, and state any version #s_
Mac OS X 10.13.5

### Behavior
* _Describe the problem_
App Crashes
* _Steps needed to reproduce the problem_
Launch App
Help -> System Info
* _Expected Results_
App does not crash. Display system info as expected. This worked in 1.40.0
* _Actual Results_
App crashes.
  * _Screenshots from __*Slic3r*__ preview are preferred_
console.log has the only interesting log message at time of crash.

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[console.log](https://github.com/prusa3d/Slic3r/files/2168758/console.log)
[Slic3r.crashlog.zip](https://github.com/prusa3d/Slic3r/files/2168760/Slic3r.crashlog.zip)

bubnikv, 2018-07-06:
> version_1.40.1-rc2/Slic3rPE-1.40.1-rc2.full-201807031453.dmg

Confirmed, thanks for the report.

enricoturri1966, 2018-07-13:
This crash should be fixed by: https://github.com/prusa3d/Slic3r/commit/b2d9877cd54bf9e4d3c13ce584b848801e3dfba6

bubnikv, 2018-07-17:
It will be fixed in 1.41.0-alpha. Closing.

closed 2018-07-17 by bubnikv