prusa3d/PrusaSlicer #1035: 1.40.1-rc2 fails to save the modified AMF settings

kursatu, 2018-07-06:
### Version
_Version of Slic3r used goes here_
1.40.1-rc2
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
1- Open an .stl and then export it as an AMF.
2-  Open that same AMF, change some setting, eg. brim width.
 The red modification indicator will appear, so will the "modified" next to the setting name.
3- Save it, it won't be effective, the red indicators will not disappear nor the "modified"
test. The app will warn about the modified settings during exit.

* _Steps needed to reproduce the problem_

[image]

[image]

[image]
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2018-07-10:
U can reproduce this issue. Thank you for report. We'll fix this as soon as possible.

bubnikv, 2018-07-17:
This will be fixed in 1.41.0-alpha. Thanks for the report. CLosing.

closed 2018-07-17 by bubnikv