prusa3d/PrusaSlicer #1036: 1.40.1-rc2 does not retain the filament names/associations while saving AMF file

kursatu, 2018-07-06:
### Version
_Version of Slic3r used goes here_
1.40.1.-rc2
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
* _Describe the problem_
The filament names are not saved and a generic name is displayed in the filament list after 
opening an AMF file.
A simple solution could be show a decorated the name in the filament list , for example
<AMFName>.<FilamentName>
and while saving the name in the AMF file 
This makes the subsequent usage of AMF practically impossible, unless you make a note of
the filament names somewhere.
You wont be able to remember with what filament you have saved the AMF file with
and load that filament onto the printer.

* _Steps needed to reproduce the problem_
1- Open an STL and it as an AMF file
2- Open the AMF file, the original filament names are replaced with generic names you
can't actually know which physical filament to load onto the printer.

If you were to use the solution I recommended above, the n the filament list would show names like
B3.zip.amf.Prusa PLA

[image]

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-07-06:
This is actually in the testing, implemented in the following branch:
https://github.com/prusa3d/Slic3r/commits/amf_activate_existing_presets

There is a release planned at the end of this month, and the branch above
should be integrated in there.

bubnikv, 2018-07-17:
This will be part of the 1.41.0-alpha. Closing.

closed 2018-07-17 by bubnikv