prusa3d/PrusaSlicer #1041: Feature Request: Preference for default units (inch, milimeter)

jexoteric, 2018-07-07:
### Version
1.40.1+

### Operating system type + version
All

### Behavior
STL files do not give units the file was created in. A file created in inches from CAD will come in the system 25.4 times too small. This can currently be addressed through manual scaling, but becomes complicated if you then want to scale the object again. Giving a preference that allows the user to change the default from MM to inches (more settings could also go here, such as centimeters), that leaves the scale shown in the interface at 100% would be helpful. 

Built into this could be a two tests, such that if an object was likely created in inches is imported, but opens in mm, the model would be very small, possibly under 10mm x 10mm, and could prompt the user, "The model you have imported is very small, did you want to switch the units to inches?" and the inverse where a model created in mm is imported with the setting of inches, and can easily be larger than the build area, in which case it could prompt, "The model is larger than the build area, did you want to switch the units to millimeters?".

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs

vertex-github, 2020-03-14:
+1.  While I can multiply by 2540%, that seems odd - it would be nicer if I could switch the project / STL import into inches and have this work nicely.  Then when I scale (normally 98-102%) its much simpler and easier to remember.

scottjanousek, 2020-09-25:
Just ran into this. Wouldn't it be relatively easy just to add a button that says scale from mm to inches and vice versa? Just a button that says mm <-> inches. Thanks.

lukasmatena, 2020-09-26:
I believe this issue is completely solved for upcoming 2.3.0 release. It is possible to convert from/to imperial, switch the object manipulation fields to show either unit and PrusaSlicer also asks when it suspects the file should be imported in inches.

I'm closing the issue.

closed 2020-09-26 by lukasmatena