prusa3d/PrusaSlicer #1045: support generation left critical component unsupported

w5xd, 2018-07-11:
### Version
1.40.0+win64 Slc3r Prusa Edition

### Operating system type + version
Windows 10

### Behavior
The support generation missed this bit of my part. It won't print

* _Steps needed to reproduce the problem_
1. Run Slic3r
2. Load the stl file I uploaded (in the zip file) sl3-missing-support.stl
3. Use the settings I attached (in the zip file) sl3-missing-support.ini
4. "Slice Now"
5.  At or about 43.8 mm into the print, the slic3r generates gcode to print something that cannot be printed as it needs support. Screen shot attached. 

That point also is about 13 hours into a 17 hour print. Oh well.

[image]

[sl3-missing-support.zip](https://github.com/prusa3d/Slic3r/files/2182532/sl3-missing-support.zip)

* _Expected Results_

Maybe I missed a parameter somewhere (I am new at this) but I cannot see why it failed to support that part of my model.
* _Actual Results_

Print fails badly.
  * _Screenshots from __*Slic3r*__ preview are preferred_

now

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2018-07-12:
Thank you for report. We will look at this in more detail if it is really a mistake. However, I think Slic3r does not create support under this part because your setting a Overhang threshold for generate supports is 45º and this part of object probably will not exceed it. When I set this value, for example to 80º (you can find better value for this model), the support was created correctly.

bubnikv, 2018-10-15:
It is better now with Sllic3r-1.41.1-rc, though the edge would not be fully printable yet as one would want a kind of cradle to be printed on both sides of the edge over a couple of layers to give it a proper support.

[image]

73, Vojtech OK1IAK, AB2ZA

closed 2026-09-29 by prusaslicer-bot[bot]