prusa3d/PrusaSlicer #1046: Feature request: jerk settings

DBa2016, 2018-07-11:
It would be very helpful to be able to modify jerk settings in the slicer, probably on the same tab as acceleration. Especially when trying to find the "optimal" jerk settings (best compromise between print speed and print quality), the need to modify it in the on-screen printer menu is very annoying.

supermerill, 2018-07-11:
You can add/edit a M205 command in the "start gcode" in your printer stenning, see https://reprap.org/wiki/G-code/fr#M205:_Advanced_settings

x0rtrunks, 2018-07-15:
Yes you can add a gcode to modify it for the whole print, but it would be nice to have similar control over jerk per move type like there is for acceleration.

DBa2016, 2018-07-16:
@supermerill adding print quality level related code into "Printer" tab means you need to maintain separate printer profiles for low- and high quality prints. Sometimes you don't care about the jerk (because there are no ghosting-threatened elements in the model) and the noise and just want to get the print fast, sometimes you want some better-looking output. Most of the quality-related stuff (like layer height, speed etc.) is no "Print settings" tab, jerk should be there, too.

supermerill, 2018-07-16:
> adding print quality level related code into "Printer" tab means you need to maintain separate printer profiles for low- and high quality prints. 

Not with conditional.
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language
you can create somethign like 
M205 X{if notes=~/.*HIGH.*/}5{else}10{endif}
or
M205 X{if perimeter_speed>30}10{else}5{endif}

bubnikv, 2018-07-21:
@DBa2016 are your needs fulfilled with the conditional G-code?

DBa2016, 2018-07-29:
> @DBa2016 are your needs fulfilled with the conditional G-code?
As a workaround until "Jerk" finds its way into the print [quality] settings, yes. With my current jerk settings (jerk limited to 3 to avoid ghosting because my Tornado is standing on vibration dampeners to reduce noise), the print takes between twice and 3 times the estimate by Slic3r. Obviously, custom jerk setting code would not resolve this issue...

closed 2026-09-29 by prusaslicer-bot[bot]