prusa3d/PrusaSlicer #1048: Explanation needed for single extruder multimaterial parameters

jimmy00784, 2018-07-11:
### Version
1.40.1

### Operating system type + version
Arch Linux 6b bit

### Behavior
[image]

This is for a multimaterial setup using a Y-splitter and a single extruder with bowden style setup. While filament parking position is self-explainatory, can someone please explain the cooling tube length and explanatory tube position? I changed the bowden setup and the heat sink is taller on the new one and now my filament kinks due to heat difference causing jams. TIA

lukasmatena, 2018-07-12:
Hi. There are tooltips that should show up after hovering above the text fields. These might explain a bit.

In short: during the unload, the filament is first rammed through the nozzle to shape its end and then quickly retracted, with its end still hot and partially melted. It is retracted to the cooling tube and moved back and forth for some time, to avoid possible sticking to the inside of the tube. Then it is retracted further to 'parking position', where it stays ready for future use.

'Cooling tube position' is the distance of the CENTER of the cooling tube from the nozzle. In other words, it is the length of the initial retraction.
'Cooling tube length' is quite self explanatory. That limits the cooling moves.
'Filament parking position' - obvious, again measured from the nozzle.

In the above example, the filament is rammed and retracted 91.5mm back. Then it is cooled while moving between 91.5-5 to 91.5+5 mm from the nozzle (not necessarily the whole distance, but certainly not further). Finally it is retracted so the filament tip ends up 140mm from the nozzle.

Did this make it more clear?

jimmy00784, 2018-07-12:
Thanks @lukasmatena 
I am still a little confused about the "Cooling tube length". I have a fan-cooled bowden style extruder setup. Does that mean the "Cooling tube length" is the length of the PTFE tube or the length of the bowden heat sink, or the bowden heat sink covered by the fan?

lukasmatena, 2018-07-12:
I don't know how your setup looks like, but maybe you can forget about the tube and focus on the functionality that I described. That's independent of any physical tube that's actually present.

Filament is retracted and moved back and forth several times before final parking. This parameter limits the 'back and forth' moves.

jimmy00784, 2018-07-12:
Sounds good. Thanks @lukasmatena .

closed 2018-07-12 by jimmy00784

phongshader, 2022-04-22:
I'm switching over to Superslicer and I'm having a hard time getting my head wrapped around getting multi extruder single nozzle settings. What is happening, with no filament swap, is the filament retracts to parking position when transitioning from part to purge tower. This seems to be a huge waste of time and also causes filament jams in my case. Is this by design? Is there a way to set this up so a simple retraction is used when there is no filament change, instead of retracting to the park position when transitioning to the purge block?
Under Printer Settings > Single Extruder MM Settings there are several settings, Cooling tube position, cooling tube length, and filament parking position. Filament parking position is self evident...where the filament retracts to when a material change happens and parks when another filament is being used. I'm not sure what the other 2 are for. I have my cooling tube position set to 35mm (aproximately the middle of the hotend heatsink) and parking position is 95mm.

phongshader, 2022-04-22:
I'm switching over to PrusaSlicer and I'm having a hard time getting my head wrapped around getting multi extruder single nozzle settings. What is happening, with no filament swap, is the filament retracts to parking position when transitioning from part to purge tower. This seems to be a huge waste of time and also causes filament jams in my case. Is this by design? Is there a way to set this up so a simple retraction is used when there is no filament change, instead of retracting to the park position when transitioning to the purge block?
Under Printer Settings > Single Extruder MM Settings there are several settings, Cooling tube position, cooling tube length, and filament parking position. Filament parking position is self evident...where the filament retracts to when a material change happens and parks when another filament is being used. I'm not sure what the other 2 are for. I have my cooling tube position set to 35mm (aproximately the middle of the hotend heatsink) and parking position is 95mm.

vovodroid, 2022-12-11:
@phongshader , did you finally succeed in finding suitable parameters? What hotend do you have - with PTFE or all metal?