prusa3d/PrusaSlicer #1049: background processing is not working any more

pintie, 2018-07-12:
### Version
_1.40.1 final 

(no Problems in all previous Version 1.3, 1.4 alphas, betas....)

### Operating system type + version
_Win 10 x64 (1803, build 17134.137) enterprise 

### Behavior
* _Describe the problem_
Slicer shuts down after 1-2 seconds when background sliceing. 
When background processing is turned off, an i hit slice now, everything works fine. If turned on, i can see all CPU threads starting for 1 second, and then slicer shuts down.

never had the issue in a previous Version. only in the final 1.40.1

* _Steps needed to reproduce the problem_
every time....
i tried it on my PC:
Win 10 x64 (1803, build 17134.137) enterprise
i7-8700k , 64 GB RAM, z370 board.

and on my Notebook
Dell XPS 15, i7-6700, 32GB Ram, same windows.

had the same issue on both machines.

_Is this a new feature request?_
no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

tried a couple of STLs, allways the same.

bubnikv, 2018-07-17:
@FidelCapo Would you please confirm that?

FidelCapo, 2018-07-17:
I have not been able to reproduce this issue with Slic3r PE 1.40.1 Win10 x64.

bubnikv, 2018-07-17:
The background processing is not stable, therefore we do not recommend to enable it to our customers. We will hopefully fix that in 1.42 release.

bubnikv, 2018-12-11:
The background processing should now work reliably. Please retest with 1.42 alpha once it is released this or the next week. Closing.

closed 2018-12-11 by bubnikv