prusa3d/PrusaSlicer #1050: Problems compiling for raspberry pi

lsellens, 2018-07-13:
version_1.40.1
on
Raspbian Stretch

I'm trying to build for octopi with no desktop. Even though I'm not running Build.pl with --gui and I'm using the cmake option SLIC3R_GUI off it's still failing to build due to lack of gui dependencies

In file included from /home/pi/Slic3r-PE/xs/src/slic3r/GUI/BitmapCache.cpp:1:0:
/home/pi/Slic3r-PE/xs/src/slic3r/GUI/BitmapCache.hpp:4:23: fatal error: wx/wxprec.h: No such file or directory
 #include <wx/wxprec.h>

bubnikv, 2018-07-17:
Thanks for the hint.

If we had excess resources, we would set up a build server for Raspbian, but as of now we are not testing compilation on R-PI.

I know there are some dependencies on the wxWigets base library for unzipping the PrusaControl archive, but those should not compile if SLIC3R_GUI is off. I am aware of one incorrect link from the libslic3r to the GUI library, that is the link to the Preset / PresetBundle structure from the AMF/3MF import/export code. We should fix that. In the meantime, pull requests are more than welcome.

Richardgriff, 2018-07-20:
lsellens

Do you have any info on you have any info on how you are building you raspberry pi version. 

I would like the same thing and I might be able to help

lsellens, 2018-07-27:
@Richardgriff Just follow the build instructions on the wiki for linux. It's pretty straight forward. Probably could also google step by step instructions if you've never used make before.

bubnikv, 2019-04-12:
You may guys try to compile the current Slic3r master on ARM.
https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

A command-line slic3r is compiled with an additional cmake switch 

`-DSLIC3R_GUI=no`

lsellens, 2019-04-13:
Do you know what dependency I'm missing? I have libboost.dev, libboost-all.dev, and libboost-chrono-dev installed already.
fatal error: boost/process.hpp: No such file or directory

bubnikv, 2019-04-13:
please post the complete build log

lsellens, 2019-04-14:
Here is the output of cmake
https://pastebin.com/XLkDDbBH
make takes a number of hours to run and its about bedtime for me. I'll post it in the morning. Thanks for your help.

lsellens, 2019-04-14:
Here is the build log.
https://pastebin.com/u8uJgdNB

bubnikv, 2019-04-14:
See the answer of 
https://stackoverflow.com/questions/1683665/where-is-boost-process

you need boost 1.64, yours is 1.62.0

We should fix the cmake to check for boost::process on Linux & OSX. It is not used on Windows.

lsellens, 2019-04-20:
Updated libboost. Getting further, but still not compiling.
cmake output: https://pastebin.com/RXShJcrn
make output: https://pastebin.com/cmRm1LpQ

vojtechkral, 2019-04-23:
@lsellens There was a bug in one of the CMakeLists, see to commit above. You can try again with the current master and hopefully it'll build. I tried building without GUI on a headless Debian machine and it worked. I needed to install these packages:

```
build-essential
pkg-config
libboost-all-dev
libtbb-dev
libcurl4-openssl-dev\
libgl1-mesa-dev
mesa-common-dev
libglu1-mesa-dev
```

lsellens, 2019-04-26:
I got to 100% finally but it's still failing with an error at the end. 
https://gist.github.com/lsellens/6fb07a3c4a459bb949db5217b8b4d4ec

lsellens, 2019-05-23:
version 2.0.0 is compiling fine now using boost 1.64
Thank you guys for all your help

closed 2019-05-23 by lsellens

bubnikv, 2019-05-23:
@lsellens you compiled without the GUI, right?
By the way, I was able to compile PrusaSlicer on Android tablet (actually an e-ink reader by Onyx Boox) using UserLAnd application, which pulls a full Debian for the ARM architecture to your Android device.

lsellens, 2019-05-23:
Yes CLI only. Just what I wanted for octoprint. Slicing is a lot faster on this new version and I really like the extra infill options. Thanks for all your hard work!

bubnikv, 2019-05-24:
thanks for heads up. maybe we should add our slicer to our preconfigured
octoprint image.

Richardgriff, 2019-05-24:
@lsellens do you have a set of instruction for compiling the new slicer 2.0

vojtechkral, 2019-05-24:
@Richardgriff it's in the [doc](https://github.com/prusa3d/PrusaSlicer/tree/master/doc) dir, it didn't change between the 2.0 and the 1.42 alphas/betas.

guebe, 2019-11-17:
Instructions for compiling prusa slicer 2.1.0 release version on raspbian or octoprint default image:

Download, compile and install boost 1.64.0
$ tar xfvz boost_1_64_0.tar.gz
$ cd boost_1_64
$ ./bootstrap.sh --with-libraries filesystem,thread,system,regex,locale,log
$ sudo ./b2 install
$ export BOOST_ROOT=/usr/local

increase CONF_SWAPSIZE to 1024
$ sudo dphys-swapfile swapoff
$ sudo vi /etc/dphys-swapfile 
$ dphys-swapfile setup
$ sudo dphys-swapfile swapon

Download, compile and install prusa-slicer 2.1.0
$ tar xfvz PrusaSlicer-version_2.1.0.tar.gz
$ cd PrusaSlicer-version_2.1.0
$ mkdir build
$ cd build/
$ cmake .. -DSLIC3R_GUI=no -DSLIC3R_PCH=no
$ echo "" > ../src/slic3r/pchheader.cpp 
$ echo "" > ../src/libslic3r/pchheader.cpp 
$ make -j1
$ sudo make install

Compiling prusaslicer uses alot of memory. Thats the reason for increasing swap file size, compiling on one core and disabling precompiled headers completely.

Now lets slice an stl file
$ export LD_LIBRARY_PATH=/usr/local/lib
$ prusa-slicer --help
$ prusa-slicer -g -s mytest.STL 

Prusaslicer is pretty fast slicing on raspberry pi 3B+!