prusa3d/PrusaSlicer #1052: Complete individual objects + increase copies + arrange = random print arrangement and order printed

GravityEcho, 2018-07-13:
### Version
1.40.1 beta win64

### Operating system type + version
Windows 10 64bit

### Behavior
Complete individual objects requires objects to be added from file (not using increase copies) and manual arranging for proper sequential printing without collisions (CR-10)
1. Add object to build plate (small object that needs to be produced in mass such as nuts and bolts.)
2. increase copies of object by pressing + or right clicking object and select increase copies to more than 2.
3. click arrange to arrange on build plate (slic3r accounts for print and head dimensions despite often arranging in a seemingly asymmetrical way depending on quantity), click slice, export the gcode.
3. Now click preview tab and tick travels to inspect the order in which the objects print. (This could also be made more clear)
This is where (more obvious with more objects) things for me go bad because there is no way to control the order in which the objects will be printed slic3r often prints in the upper left quadrant then beneath that (y axis) then it tries to go back up and to the right and KNOCKS OFF the first item with the X gantry.
It should never try to go back up it should complete items along the X THEN go down left and print another row along X.
* _Expected Results_
Print individual objects without hitting objects with print head or X gantry
* _Actual Results_
Depending on print quantity and size can still crash because user is not given a few options to help control the print order.
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]
It may be hard to see the travel lines in this image but the problem is that the increase copies and arrange method has made slicer print the middle object in the Y row of 3 first then up to the top item in the Y and then over to the one by itself and then the bottom one in the Y. It just doesn't make any sense to me to have it arrange them so strangely and print in such a haphazard order certainly will cause collision problems.

What I would like "Increase Copies" and "Arrange" to do is more like the following:
[image]
In this image I added each part by dragging and dropping the file into Slic3r 4 times and then manually arranging those objects on the build plate (a snap feature would be helpful, forgive me if I am unaware of it) to make it print top left, top right, then bottom left and finally bottom right. I have a print running with 9 of these using this method and it is working. Which is excellent, I love Slic3r its the best!

_Is this a new feature request?_
Yes
Please add more options for the "arrange" feature such as:
align along X axis then Y (Arrange along X until out of room then down Y and along X again)
align along Y axis then X (Arrange along Y until out of room then across X and along Y again)
And/Or allow user to control the order in which the objects are printed/arranged.

Thank you so much for this wonderful software!

zerg960, 2018-07-22:
Preventing collisions is a bit harder of a problem than just arranging the objects in a certain order, once you consider features like crash detection & filament runout:

[image]

If crash detection is triggered, the extruder auto-homes. The extruder heads to 0 on the X axis, then heads to 0 on the Y axis (because it's auto homing). Printing in the order above would crash the extruder into already completed prints in the process.

Then the extruder heads back to where it was. (I had an experience where the nozzle, as it was heading back crashed into an already completed 10mm-high print, ruining it, the position of one of the Z-axis rods, and the heater body too.. and it triggered crash detection again. That was the last time I used this feature.)

At least on the MK3, the extruder also goes to a different corner if the filament runout sensor is triggered, or the user pauses the print.

All those thorny problems aside, it's possible for an algorithm to check if a collision with the extruder is possible by drawing a 2D line from the 3 corners and seeing if it intersects with any corner of the bounding boxes of any already-completed print (plus a margin equal to the extruder's body size). As long as the parts aren't taller than the distance between the X-axis guide rod and the nozzle, collisions with the gantry aren't possible.

dartrax, 2018-07-23:
#810

dartrax, 2018-08-21:
@mjsadaway, could you please explain how and why that would prevent collisions, i.e. collisions caused by filament runout detection, and what you mean by extruder cable?

mjsadaway, 2018-08-22:
It wouldn't prevent collisions caused by filament runout or crash detection, both of which cause the MK3 to auto home. It should help in the case where completed parts near the back of the build plate are knocked over by the large bundle of wires sticking out the back of the extruder.

Yes, it would be nice to solve the problem in every case, but I would be happy if I could rely on the "complete individual objects" function most of the time, when the filament didn't run out and a crash didn't occur.

chrschn, 2019-03-05:
I think giving uses control about the print order (see #1338) would at least allow one to manually handle complicated use-cases. 

My use-case would be printing multiple copies of the same object in different colors using an MMU. The printer should complete one object in one color, then do a tool change, then print a manual wipe object before starting to print the next object with the new tool. This would be possible to achieve if I could manually set the print order of the objects (in particular printing the wipe object before the actual object).

bubnikv, 2020-02-04:
The upcoming PrusaSlicer 2.2.0-alpha4 will print sequentially in the order displayed at the object list. Also the object list will support drag & drop to reorder.

luismasg, 2020-03-06:
I'm doing nintendo yoshis and i wanted to print 4 sets of shoes different color, the height es about .8 cm so the support rod is not an issue.
seems like its ready to print. but i see no wipe tower.
I would expect for the  first layer to be the color transition. I might just try it . I'm doing the soles right now.
Is this 'complete individual objects'  feature considered for MMU ?
it would vastly reduce filament swap from 30+ to  4  
does anyone if something like this is on the roadmap ?
or is it possible ? 
[image]

bubnikv, 2020-03-07:
Wipe tower is not supported for sequential print.

freakydude, 2023-04-14:
Hi, 

I was posting a short pseudo algorithm to improve the automatic arrange in the linked issue above. 

I was rethinking about it and like to share the my idea in some diagrams. I am sure that others had similar ideas before me, I'm not well informed about that - feel free to participate.

In my opinion there are absolut basic operations we need to support anyway. This basic ideas should be easy to implement and would help for a lot of use cases already. 

**Basic Idea:**

First of all, all we need is the *x and y distance (Xd,Yd) from the nozzle tip to the front left corner of the extruder*. There is no need for a radius or clearance height are at all.

The following diagram shows a cartesion printer from the top. All we need to know is the red rectangle, or more specific the xy-distance from the nozzle tip to the front left corner. 

[image]

This enables us to make some easy calculations for a collision free printing in x or y or combined xy sequential printing direction. In addtion the the xy-distance we need the maximum printing area of each object to print. That should be well known already, because the normal arrangemant algorithm is doing very well here.

So for printing in x-direction:
- Start at "minimum" x.
- Position/Print the object. 
- Arrange the next object to print by shifting it by x-dimension of the object before + the x-distance (Xd) of the nozzle-tip to extruder left
- Repeat for the next object

Print in x-direction:
[image]

The same as for x-direction, but y.

Print in y-direction:
[image]

The same as for x-direction, but x and y can be variated at will. 

Combine print in x- and y-direction:

This is the reason, why the basic steps are needed and this is also the point where in future we can think about further optimizations. For example, sorting by x,y object size, modelling more detailed the xyz clearnce of extruder/xaxies and object height. Everythink to use the space of the heater bed more efficiently. I have also some ideas, maybe I add that later on. 
[image]

**Conclusion:**
- Clearance Height and Radius are irrelevant first
- X and Y Distance from Front-Left Extruder to Nozzle Tip are the needed variables
- Max  x-y size of the objects to print are needed (and should be already there, because normal arrangement algorithm is doing very well)

What are you thinking about it?
Any hints, where is the place to implementing this - or someone who like to try that?

With best regards,
freakyDude

891paul, 2023-05-17:
Look at 
https://github.com/prusa3d/PrusaSlicer/issues/10552#issue-1708902322
#10552

lukasmatena, 2025-02-18:
Arrange and crash detection for sequential printing was improved in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). I believe it solves this issue. Closing.

closed 2025-02-18 by lukasmatena

freakydude, 2025-02-18:
@lukasmatena This is great news today. Thanks for finally looking into this. I am trying my first sequential print with Voron and your new algorithm. By the way, I am using a clearance of height 42, radius 52 for Voron 2.4 Stealthburner measured according to the official Voron assembly model.

Thank you

freakydude, 2025-02-18:
I can confirm - on my first attempt, both parts were printed one after the other

**But I have a question:**
It's still unclear to me whether the "clearance distance radius" includes the x-carriage rods and the back of the tool head - or just the largest dimension of the tool head itself? Would be nice to clarify that in documentation to prevent things from breaking.

lukasmatena, 2025-02-18:
The two parameters mean exactly what is shown in [this picture](https://github.com/user-attachments/assets/713399e4-f914-488d-89f4-9065813f35f9) in the change log, and what you see in the preview. The extruder is approximated by a cylinder of radius `r` and a prism of y dimension equal to `2r` and lifted `h` above the bed.

freakydude, 2025-02-18:
Fine that was exactly what I did understand, but was not sure about. 
Good to know that the picture is well thought out by you. Thanks again.

If there is a way to contribute the voron default stealthburner model in future, just let me know how.

Best regards