prusa3d/PrusaSlicer #1053: Strange slicing behaviour

ahjulstad, 2018-07-16:
### Version
Version 1.40.1+win64

### Operating system type + version
Windows 10
[stl_and_config.zip](https://github.com/prusa3d/Slic3r/files/2198984/stl_and_config.zip)

### Behavior
When slicing the attached STL, a closed layer is generated in an area where there are gaps in the STL. 

[image]

[image]

If I rotate slightly (2 degress around X axis), there is 'space' (only support material)

[image]

STL and config attached. The STL is generated by Fusion 360, and must be rotated 90 degrees first.

bubnikv, 2018-07-17:
@FidelCapo Would you please test that? Will fixing the model by NetFabb help?

zerg960, 2018-07-22:
Tested it. I can confirm that the STL does have odd slicing and that rotating by 2 more degrees around the X axis "fixes" it.

The STL is non-manifold (566 reversed faces, 20 backwards edges). And fixing the model by NetFabb does help.

bubnikv, 2018-07-23:
@FidelCapo also confirmed that fixing the object in Netfabb helps (you can call the Netfabb service directly from Slic3r on Windows 10).

bubnikv, 2018-10-05:
The fix will be released with 1.41.1-rc. Closing.

closed 2018-10-05 by bubnikv