prusa3d/PrusaSlicer #1054: Unwanted solid infillsurface in the middle of part

eowindel, 2018-07-17:
### Version
1.40.1+w64

### Operating system type + version
W10 64bitss

### Behavior
Slic3r add solid unwanted infill in the print. The unsure vertical wall thickness is uncheck.

For exemple S3D on the left (what i want) and slic3r on the right. I tried also is Cura without problems.
[image]

Other one with a larger print
[image]

#### STL/Config (.ZIP) where problem occurs
[STL-config.zip](https://github.com/prusa3d/Slic3r/files/2200337/STL-config.zip)

Anima1000000, 2019-02-09:
### Version

1.42.0 alpha5 MacOS 10.13.6

Same problem occurring for me. It works fine with 2 or more perimeters but with a single perimeter I am getting a lot of unwanted solid infill added in between the infill.

With ensure vertical wall thickness enabled I am actually getting less of this solid infill noise then when enabled.

bubnikv, 2019-02-09:
please provide screenshots and a 3mf file as requested in the issue entry
form.

Anima1000000, 2019-02-09:
Thanks for looking into this. Notice that with 2 perimeters this issue doesn't occur.

[20190209_SampleB.3mf.zip](https://github.com/prusa3d/Slic3r/files/2847987/20190209_SampleB.3mf.zip)
[image]

supermerill, 2019-02-12:
> Notice that with 2 perimeters this issue doesn't occur.

Probably because the second perimeter remove the top surface. No top surface -> no solid surface below.
Note that any top surface is grown by 3mm in all direction.

Anima1000000, 2019-02-12:
Hi Merill, have you had the chance to look at the file? These solid infills are throughout the entire object not just under the top layer. With two perimeters it doesn't remove the toplayer.  Setting the top layers to 0 doesn't remove these solid infill artefacts. However disabling both top and bottom layers does fix it.

Any idea what might be causing this or is this infill intented for 1 perimeter? If so any chance to disable it?

supermerill, 2019-02-12:
see https://github.com/prusa3d/Slic3r/issues/668

Anima1000000, 2019-02-13:
I tried your build, it's has some really nice options. Would be nice to have a MacOS build aswell.

However I have the same problem there, I have tried all the options. I think it sees cavities inside the object as top and bottom surfaces, that must cause the unwanted infill.

Anima1000000, 2019-02-13:
Found a workaround by using infill modifiers.

tsawher, 2019-05-27:
Adding to this issue with another example. This causes a lot of time and material waste.

### Version
PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Windows 10 Pro 10.0.17134

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 mk3
3.7.1-2266

### Behavior
PrusaSlicer will put seemingly needless solid infill and top solid infill in the middle of prints. This causes significant increases in printing time and material usage (Cura prints the same model with the same speed settings in nearly half the time).

This is not a new feature request.

Attached is model, project, and config ini.

The problem is most evident starting at layer 74 on up. You will see the seemingly random parts of top solid infill (and some bridge infill). This project takes 6 hours 52 minutes in PrusaSlicer, but in Cura it takes less than 4 hours when using the same speed settings. The biggest difference is that Cura doesn't slice in all of the extra infill.

[image]

#### Project File (.3MF) where problem occurs
[Bug report.zip](https://github.com/prusa3d/PrusaSlicer/files/3223800/Bug.report.zip)

jltx1, 2019-10-02:
Is this getting fixed?   I see this rampantly in 2.1

ftheriachab, 2019-11-17:
I found a fix, just turn off "Ensure vertical shell thickness" in the Print settings tab, under Layer and Perimeter

supermerill, 2019-11-20:
@ftheriachab no, it's not. prusaslicer increase the area of solid infill to ensure it's going well over perimeters, and this setting isn't exposed. "Ensure vertical shell thickness" is an other thing that also add some solid area, but it's not on the same scale nor in the same areas.

robschwieb, 2019-12-26:
Is there any update to this? There has to be someway to disable this "feature" other than limiting either top or bottom layers to 0. This is the only slicer I've seen do this and it's very annoying for certain prints.

Buzzz57, 2020-02-07:
Same issue here. Very annoying

djen-k, 2020-02-28:
I am also encountering this issue a lot! Would be very nice to have it fixed in the next update!

VanessaE, 2020-03-17:
Still an issue in 2.2.0-rc4. :confused:

CodeCactus, 2020-04-08:
Not sure if this will help all of your specific use cases, but if you decrease the number of top/bottom solid layers, you will have less of this, or none at all. I came to this thread with a similar issue, and doing that fixed mine, best of luck with yours!

haleypearse, 2020-04-13:
This is how I mainly use modifiers. I set the overall top/bottom layers to 0 or 1 and modify the top and bottom to the number I actually want. Ridiculous.

MadMikey32, 2020-04-16:
Hi Guys,
Just recently notice this unwanted Solid infill in random Layer Height. 
So i stated to research online to see if someone had this issue. Didnt find a solution.
Started to play with setting on Slic3r.

So I found the solution :)

I'm currently using PrusaSlicer 2.2.0

So here is a before & after

Layer & Perimeters setting before
[image]
[image]

Layer & Perimeters Settings After
[image]
[image]

Not sure why is this worked.
The settings are for Horizontal Shells.
The Top & Bottom Solid layers need to match.

Hope this helps you guys.

Jannes351, 2020-04-18:
Hi guys

Was struggling with this too, especially on small objects. Found a solution!
[https://help.prusa3d.com/en/article/infill_42](https://help.prusa3d.com/en/article/infill_42)

The setting that forces this on small objects (and large ones with small surfaces) is Solid infill threshold area, set to 0 to disable.
[image]

bluesign2k, 2020-05-02:
@Jannes351 Only just got the chance to look at this, but changing this value seems to have resolved my issue. Thanks! 👍

nospoon, 2020-05-10:
This option is disabled by default and it still generates those solid infills for even with it disabled. Setting top/bottom layers also seems to have no effect. Have tried everything now and still can't get rid of those solid infills. I noticed it gets much worse after lowering perimeters to 1. 
This totally messes up my flex objects properties.

haleypearse, 2020-05-11:
It's crazy this bug has been around for years, and has such a big effect. I print at lower layer heights just to avoid it, or use modifier meshes in the middle of the part to lower top/bottom layers to 1 or 0 where there are no real top or bottom surfaces. The best reason not to use (prusa)slic3r (but (prusa)slic3r is still way better than the competition). I want to learn to code well enough to fix it.

Mat172, 2020-05-18:
Same problem here, i´ve tried everything to get rid of it.

tommck, 2020-10-08:
I just noticed this problem with a print I'm doing.. 0% fill density, yet I get solid infill in the slicing:

[image]

supermerill, 2020-10-13:
The most likely: That's because there isn't enough perimeters to have at least bottom_solid_layers at these spots, so it has to add more plastic.

VanessaE, 2020-10-13:
That would be fine IF it were adding more *perimeters* to fill it out.  But instead, it's adding solid infill, which will *always* end up performing worse.

Still, that highlights a logic issue: you should not compare to number of *layers*, you need to compare *thicknesses* instead.

supermerill, 2020-10-13:
but adding more perimeters means also adding a perimeter where it isn't needed, and never using infill as we are adding perimeter instead...
It's not easy.
imo, using a special infill pattern (kind of a concentric with gapfill) if the infill is "solid & inside" and lower than a threshold in width and area ; may be the desirable solution.

haleypearse, 2020-10-17:
Off-topic but since you mention it, is there any thought of enabling partial perimeters? It occurred to me that full loop perimeters are too great a limitation. For instance when you use a modifier mesh for a different number of perimeters, it creates a double-wide dividing down the middle of the part, which is in general so far from what you want that it's funny.

fcore117, 2020-10-17:
Anyone knows how to get rid of that unwanted goo in 2.3.0 alpha1
[image]

fcore117, 2020-10-18:
Vase mode is only option that disables this but prevents adding more layer and if adding layer then this infill blobs are back.

stefanct, 2020-10-27:
Can everybody please stop propagating the following settings as a fix/workaround - they are clearly not (the only culprits) as can be read above.

* Layers and perimeters
 - Minimum shell thickness
 - Extra perimeters if needed
 - Detect thin walls
 - Ensure vertical shell thickness
* Infill
 - Solid infill every (not mentioned yet, but I have tested it as well)
 - Only infill where needed
 - Solid infill threshold area
 - Fill angle (not mentioned yet, but I have tested it as well)

It seems to me like a bug in some heuristics related to sloped perimeters that somehow is related to the number of perimeters (higher is better) and horizontal shells (lower is better). In my very limited testing I have found that increasing the number of perimeters from 1 to 3 reduces the estimated printing time from 23 to 19 minutes for a small, irregular object with many planar slopes). This is with all other settings at non-interfering/optimal values. Thus you can try mitigating by adding perimeters but this is clearly a bug/missing profile knob.

Freijan, 2020-11-05:
I'm so sick and tired of this bug. I just tried 2.30 alpha 3, and it looked to be even worse than 2.2 for my prints. None of the mentioned workarounds/fixes works for me.

The only thing that seems to lower the amount of wild appearing solid infill noddles is to set the infill to 0 %.

[image]
[image]
[image]
[image]

VanessaE, 2020-12-07:
So what gives?  Is this ever going to get fixed?

MauriMo94, 2021-04-15:
I had the same issue with cura. I solved it by setting the bottom layers for the whole model to 0. Then created a box with overlapping mesh settings to 1.686mm for the bottom.
[image]

Phil1988, 2021-04-16:
+1 for this.
I am also having frequently trouble with this.

baselsw, 2021-04-16:
I just followed the advice from @MadMikey32 . I temporarly solved the problem by turning off/unchecking "Ensure vertical shell thickness" and setting the number of Top and bottom solid layers to 4 (I think they have to be equal for this to work).

TriangularBubble, 2021-04-30:
I found a _working_ workaround as everything above didn't work for me:

In PrusaSlicer 2.3.0 I did following-> use a modifier box and blow it up to the size that the bottom and the top are just looking out. Then set the top and bottom shells to 0 in the modifier box (in layers and perimeters). See pics. As the bottom and top are not inside the modifier box you'll still have those for the model. 

I guess its adding layers as its not detecting exactly where the end top and starting bottom is if there is any small opening. 

Happy printing :)

[image]

[image]

fcore117, 2021-04-30:
Is this still a thing? 2021 is year. Perhaps PrusaSlicer should integrate CuraEngine too as alternative selectable choice. Different engines could solve different user cases and needs.

tommck, 2021-05-14:
All, I just ran into this again with a model where I was doing 0% infill and multiple perimeters. 

What wound up working for me was: 

Disable "ensure vertical shell thickness" and enable "detect bridging perimeters", like this (2 orange settings):
[image]

Protxch, 2021-05-14:
Same here on this Tolerance Gauge. Turning off "ensure vertical shell thickness" and enabling "detect bridging perimeters" seemed to help a little but not much 
[image]

Phil1988, 2021-05-15:
did you try to have equal bottom and top thicknesses?
For some, this was a workaround, but I also have had models where this didnt worked.
This is pretty sad because PrusaSlicer is really great otherwise.

GlennovitS, 2021-06-29:
How can this still be an issue after so many years? I guess that when you set top / bottom layers, you intend to only adjust the top and bottom layers, not every single layer with angles through the model. The suggestion above with a modifier box to select the whole model expect the top and bottom, and then set top/bottom to 0 layers do actually help. But that just seems wrong that we have to do that in every model with angles.. Please fix it Prusa team

robschwieb, 2021-08-23:
Looks like we missed commenting on this in July. I got the August comment covered...

supermerill, 2021-08-23:
my current fix for this issue:
supermerill/superslicer@3153d9612e8279404a6d886c12abe2781f4761dd

fcore117, 2021-12-03:
Anyone knows to report that is this still a problem on 2.4.x releases? or they solved it once and for all?(option) thanks.

Phil1988, 2021-12-04:
For me this is still unsolved.

I switched to SuperSlicer because @supermerill solved it there 👍

ArrowRaider, 2021-12-10:
Vertical Shell Thickness on or off makes no difference.  I am using SuperSlicer instead which doesn't have this problem.

fcore117, 2021-12-10:
I still think why PrusaSlicer have not still given option for this or at least give selectable backend engines to slice like CuraEngine which really shines in geometry and structural beauty generation, but UI itself is ultra slow and uses massive bloated UI programming language.

bubnikv, 2021-12-11:
We plan to work on it for PrusaSlicer 2.5, it is high on our list. Unfortunately it did not make it to PrusaSlicer 2.4.

jozkoxoxo, 2022-06-29:
> We plan to work on it for PrusaSlicer 2.5, it is high on our list. Unfortunately it did not make it to PrusaSlicer 2.4.

as of today, PS 2.5 alpha2 is out, but I don't see anything related to this issue in the changelog
could you please clarify if there is something new?

bubnikv, 2022-06-29:
It is on our radar, but it did not make it into PrusaSlicer 2.5.0.

user1419, 2022-08-07:
That problem is really annoying for some prints. A fix would be greatly appreciated, since prusaslicer is really good besides that probmem.

napter, 2022-10-05:
This is still an issue in 2.5, none of the workarounds work - at least not if you use one perimeter. It adds a lot of extra plastic and time to prints. @supermerill fixed this in a while ago, can it be ported over?

fcore117, 2022-10-05:
CuraEngine would be solution if it cannot be fixed. Seems that even in 2023-2024 this thread still might be open.

n8bot, 2022-10-06:
The problem is not easy to fix. There are two algorithms at play. Both of them kinda do the same thing. If you choose "Ensure vertical shell thickness" it does one algorithm, if you unselected it chooses another.

The problem is, you can't just disable both algorithms. One of them is required to one to sufficiently prepare the file for printing. Prusa research did not, AFAIK, code it this way. Their additions tend to be much more discrete. These legacy implementations are very intertwined with the rest of the process.

While it might be tempting to just go and rip them out, and then solve the problems as they come -- that can lead to immense unforeseen consequences and end up taking more time than would be taken careful planning an entirely new algorithm. Whack-a-mole-in-production is probably not a game Prusa is looking to spend more time playing.

So... while, yeah, it sucks that it's still like this... I've seen the code. I've played with it a lot. It's NOT a simple task to remove.

napter, 2022-10-06:
@n8bot I haven't looked at the code but @supermerill already fixed this in the SuperSlicer fork of PrusaSlicer. Perhaps that code can be ported or he can explain how he did it.

n8bot, 2022-10-06:
Yes I do see that now. It does seem like a decent workaround until the underlying problem is fixed.

nnarain, 2023-01-12:
Hello. The title of [this issue](https://github.com/prusa3d/PrusaSlicer/issues/9245), created by @dehmlowm describes my issue better. But it seems all roads lead here..

PrusaSlicer tends to create these small squiggly paths that cause excessive vibration while printing on my Ender 3. I've tried this at various speeds and I don't typically deviate from the stock settings too much. This often results in layer shift for me.

[image]

@tommck's settings above seem to increase the size of the squiggles which I assume will (hopefully) reduces the mechanical vibration. But I have yet to try it.

Does anyone have any advice specifically for reducing the vibration?

dehmlowm, 2023-02-28:
Thanks for referencing my post! I _think_ that issue 223 is the original issue.I will make the same offer here that I will Venmo anyone who fix this bug $500 US. I included a straight forward example of the files here:
https://github.com/prusa3d/PrusaSlicer/issues/9245

[2 more replies]

closed 2024-11-21 by lukasmatena