prusa3d/PrusaSlicer #1055: Can't flash printer firmware unless opening Slic3r as Root user

agarod-git, 2018-07-17:
**Version**
1.40.1+linux64 and older

**Operating system type + version**
Ubuntu 18.04

**Behavior**
If I open Slic3r as normal user and try to flash printer firmware, it can't be done because permission to open Serial Port is necessary:

[image]

When opening Slic3r as Root user, flashing is done successfully.

**Expected Results**
Be able to flash printer firmware by opening Slic3r as normal user.

**Is this a new feature request?**
Yes

Thank you.

bubnikv, 2018-07-17:
Sorry, but it is your responsibility to allow non-root access to the serial ports.

@vojtechkral Please add a test & help message to guide the user on how to allow non-root access to the serial ports.

agarod-git, 2018-07-18:
@bubnikv Thanks for your prompt response. It directed me on the right path.

I connect my laptop via USB cable to the printer.
I found this [answer](https://unix.stackexchange.com/questions/14354/read-write-to-a-serial-port-without-root#14363) from StackExchange.

With the printer connected, I checked that the serial port is owned by user **root** and group **dialout**.
So the command provided in the answer works and you have to run it as superuser:

`sudo usermod -a -G dialout <my_user_name>`

**Note**: don't forget to **log out** and **log in** after running the command.

Thanks guys for your great work! :)

zerg960, 2018-07-22:
On Windows 10, I encountered an access denied issue flashing firmware earlier today. Running as administrator fixed it (but I had to re-plug the cable in one more time, since even being admin still gave an access denied error).

Vedrillan, 2020-05-25:
The solution by ghost is not advisable as it could give rootless access to more than just the 3D printer.

This is actually a simple matter of udev rules, which should be officially documented to be honest, as the majority of linux users will have this issue and a unnecessary hard time to deal with it.

The solution is to add the custom udev rules by running the 2 following commands in a terminal:
```
echo 'SUBSYSTEMS=="usb", ATTRS{idVendor}=="2c99" TAG+="uaccess", TAG+="udev-acl"' | sudo tee /etc/udev/rules.d/20-prusa.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
```

justindelliott, 2020-05-25:
Thanks @Vedrillan , that fixed it for me on Linux Min 19 Tara (Ubuntu bionic)!

Agreed that it should be officially documented for Linux users.

numbermess, 2022-01-23:
Thanks @Vedrillan! That worked great. I was able to update my printer using your comment in 2022. 

> avrdude-slic3r done.  Thank you.

peepls, 2022-08-11:
This worked for me too, running Linux Mint 20.3
Isn't it possible to change these settings automatically during installation?

egg303, 2023-05-15:
Thanks @Vedrillan, works on SteamOS 3.4.4

closed 2026-09-29 by prusaslicer-bot[bot]