prusa3d/PrusaSlicer #1056: Temperature extruder

Melandr, 2018-07-17:
### Version
_Version of Slic3r used goes here_
[image]

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 7 SP1

### Behavior
when slicing, the temperature is set to 200, and not the one that is set in the settings
[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

zerg960, 2018-07-21:
Can you export your config?

Melandr, 2018-07-21:
Yes, export config 24.07.2018. because now on vacation

21.07.2018 18:56 пользователь "kreatious.net" <notifications@github.com>
написал:

Can you export your config?

—
<https://github.com/prusa3d/Slic3r/issues/1056#issuecomment-406805780>, or mute
the thread

Melandr, 2018-07-24:
[config_slic3r.zip](https://github.com/prusa3d/Slic3r/files/2224488/config_slic3r.zip)
Export config slic3r prusa edition

bubnikv, 2018-07-26:
Would you please update your AMF or 3mf, and the generated G-code? Thanks.

FidelCapo, 2018-07-26:
I tried to reproduce this issue, but I did not. Can you please send the gcode in which the error is displayed and all steps needs to reproduce this? Thank you.

bubnikv, 2018-12-11:
No response, closing.

closed 2018-12-11 by bubnikv