prusa3d/PrusaSlicer #1057: 'Preview' and 'Layers' tab display issue on Mac

echeng622, 2018-07-18:
### Version
1.40.1+

### Operating system type + version
MacOS 10.13.6 (17G65)

### Behavior
1. Launch fresh download of Slic3r from prusa3d.com
1. Load STL 
2. Object shows up as expected in '3D' & '2D' tab
3. Navigate to 'Preview' tab and a empty printer bed is displayed. Object does not appear
4. Navigate to 'Layers' tab, and a white window is displayed.

lukasmatena, 2018-07-18:
Preview and Layers tab show results of slicing. Unless you don't actually slice the model, there is nothing to show. Does the issue manifest even if you slice the STL after loading it?

echeng622, 2018-07-18:
@lukasmatena Thanks that resolved it! Guess thats a newer feature.

closed 2018-07-18 by echeng622