prusa3d/PrusaSlicer #1061: [Linux] Unlisted dependancy - libnsl

Kolous, 2018-07-20:
### Version
Slic3rPE-1.40.1.linux64-full-201807051330.AppImage

### Operating system type + version
Fedora 28

### Behavior
Appimage crashes with following error:

/tmp/.mount_Slic3rCy9AOD/usr/bin/bin/perl5.22.0: error while loading shared libraries: libnsl.so.1: cannot open shared object file: No such file or directory

Installing libnsl satisfies the dependency.

vojtechkral, 2018-07-20:
This is a new issue that has arisen recently with AppImage (see: https://github.com/AppImage/AppImages/issues/336) because `libnsl` used to come with `glibc` and/or was installed by default on systems but apparently that's not the case anymore.
It seems `libnsl` is on the [AppImage bundleed library exclude lists](https://github.com/AppImage/AppImages/blob/master/excludelist) but might need to be removed.
I don't think there is any consensus yet over at AppImage as to what should be done about this.

bubnikv, 2019-04-13:
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1 does not use perl, so this should be fixed.

closed 2019-04-13 by bubnikv