prusa3d/PrusaSlicer #1062: [Linux] Running Slic3r as root improvement

Kolous, 2018-07-20:
### Version
Slic3rPE-1.40.1.linux64-full-201807051330.AppImage

### Operating system type + version
Fedora 28 x64
Linux shadow 4.17.2-200.fc28.x86_64 #1 SMP Mon Jun 18 20:09:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
Firmware flashing fails with permission denided error while opening the port therefore root permissions are required. When trying to run app as a root (sudo, su, su -) i get following error:

```
17:14:47: Error: Unable to initialize GTK+, is DISPLAY set properly?
Failed to initialize wxWidgets at /tmp/.mount_Slic3rCkuZxl/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/2DBed.pm line 10.
Compilation failed in require at /tmp/.mount_Slic3rCkuZxl/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/2DBed.pm line 10.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rCkuZxl/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/2DBed.pm line 10.
Compilation failed in require at /tmp/.mount_Slic3rCkuZxl/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI.pm line 9.
BEGIN failed--compilation aborted at /tmp/.mount_Slic3rCkuZxl/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI.pm line 9.
Compilation failed in require at (eval 78) line 1.
```

To run app as root I use:
```
xhost +local:
sudo Slic3r......
```
But in this case the user configuration (user printer profile) is not directly accessible. I propose to show password request dialogue when updating the firmware and run updater in separate thread with root priviledges.

bubnikv, 2018-07-20:
We do not plan to separate the firmware updater into its own process, therefore we cannot implement your request.

We recommend to make the serial port accessible to everybody
```sudo usermod -a -G dialout <my_user_name>```

as described here:
https://github.com/prusa3d/Slic3r/issues/1055

We will also make a dialog pop up explaining to the user what needs to be done.

Closing this issue and keeping https://github.com/prusa3d/Slic3r/issues/1055 open.

closed 2018-07-20 by bubnikv

tracernz, 2018-07-20:
To add, you should _never_ run GUI apps as root, the exposure is too great, and there is always a better way (even setcap is a lot better).