prusa3d/PrusaSlicer #1063: Load Config Bundle Error: Duplicate Section Name

pikecity, 2018-07-20:
### Version
_Version of Slic3r used goes here_
Slic3r Prusa Edition

_Use `About->About Slic3r` for release versions_
1.40.1+

### Operating system type + version
Apple MacOS 10.13.6

### Behavior
* _Describe the problem_
After saving a Bundle Configuration File attempts to reload it results in a "Duplicate section name" error.  

[image]

This issue occurs before Adding an STL file or after. But I've included and STL file nonetheless.

Inspection of the ini file confirms a duplicate section.  In the attached example bundle ini file it is near the end and is called: [printer:Original Prusa is MK3 0.6 nozzle]

Removal of the section results in same error when I attempt to load said file and re-inspection shows that it has been inserted back into the file: but why & where is this coming from?

Note that I've looked for duplicate individual Filament/Print/Printer Settings, but have not found any. 

Considering that I have 1 bundle that I saved recently that works, this may not be a BUG in Slic3r and is probably a PEBKAC error - but what did I do?

* _Steps needed to reproduce the problem_
Saving then using the Load Config Bundle function creates problem.

* _Expected Results_
Load bundle and associated configuration files.

* _Actual Results_
See error message above...

 * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No

#### STL/Config (.gz) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[config.ini.gz](https://github.com/prusa3d/Slic3r/files/2215661/config.ini.gz)
[PETG PRUSA BLK 072018.ini.gz](https://github.com/prusa3d/Slic3r/files/2215662/PETG.PRUSA.BLK.072018.ini.gz)
[ExampleSTL.stl.gz](https://github.com/prusa3d/Slic3r/files/2215663/ExampleSTL.stl.gz)

bubnikv, 2018-08-02:
@vojtechkral would you please look into this issue?

Patrik-svobodik, 2018-12-17:
Happens for me as well, I cannot import any profiles

bubnikv, 2019-04-13:
I hope these issues are now a history with the https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

closed 2019-04-13 by bubnikv