prusa3d/PrusaSlicer #1064: Mission section in print, yet exists in preview

Pathogenius, 2018-07-20:
### Version
1.40.1-rc2+win64

### Operating system type + version
Windows 8.1 (fully updated)

### Behavior
Missing section in print (second layer), yet exists in Slic3r layers preview:

[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[files.zip](https://github.com/prusa3d/Slic3r/files/2215672/files.zip)

Pathogenius, 2018-07-20:
I tried twice, each with different Z-rotation in Slic3r.

zerg960, 2018-07-21:
Can you upload the g-code? Here's what I get with your settings (I'm on 1.40.1+x64):
[Turntable_Split_5pcs_1.zip](https://github.com/prusa3d/Slic3r/files/2216301/Turntable_Split_5pcs_1.zip)

Pathogenius, 2018-07-21:
[Turntable_Split_5pcs_1.zip](https://github.com/prusa3d/Slic3r/files/2216346/Turntable_Split_5pcs_1.zip)

Pathogenius, 2018-07-21:
and I happened to notice its doing the same thing, not printing sections of the first layers, the image is the 'cube-40-40-40.stl' provided by Prusa, looking through the bottom as I didn't get any shots during print, but the artifact is obvious.

[image]

zerg960, 2018-07-22:
I analyzed the .gcode you provided using diff, and a few different g-code analyzers. The toolpath in the G-code does not have a gap, and the extrusion rate at the beginning of layer 2's infill isn't odd, ruling out a slicer bug.

Interestingly, I ran your .gcode on my MK3 and had that **same gap** on the 2nd layer (you had lower temperatures, a maximum print speed of 100mm/s, and 10% gyroid infill). I also had to lower my live Z adjustment (squishing more) by 0.03mm before I could get a first layer that sticks.

Then, I sliced the turntable .stl and printed it with the stock 0.20mm profile (the only change I made was to use 10% gyroid infill), with the stock filament settings, and I did not have that gap. A diff showed that the only .gcode changes was on lines stating: `G1 F<some number>`.

Next, I sliced the .stl again, used 10% gyroid infill and your slower print speed settings (I only kept the higher temperatures), printed it, and again, no gap. A diff proved that the only .gcode changes were to the temperatures.

And just to confirm it's temperature related, I took that .gcode and ran it again, manually adjusting the temperature of the first layer to 200/50 and the 2nd layer to 195/50 using the menus. And the gap came back. The combination of the 2nd layer fan and the 5 degree temperature drop caused the nozzle temperature to drop way below 190. I didn't hear any extruder clicking, but I did see that the nozzle continued to move around. I have a strong suspicion that the firmware has a feature that prevents the E-axis from turning if the nozzle temperature is below 190.

Try adjusting your printing temperatures back to 215/60.

Hope this helps.

Pathogenius, 2018-07-22:
Interesting! I had initially lowered the temp to solve an issue that I later suspected was not in due to temperature, but rather bed adhesion. I later learned about using soap to clean the bed and increase adhesion, but did not yet attempt to raise temperatures as things were working for the limited set of prints I did thereafter. I will certainly start from square one now with purely default Prusa profiles, and surely continue to develop my printing skills. Thank-you zerg90 for revealing the problem and putting me back on the right track! I will close this issue now and if for some reason correcting temperatures does not solve things as it did on your end I will reopen in firmware sub. Cheers!

closed 2018-07-22 by Pathogenius