prusa3d/PrusaSlicer #1065: XY Size Compensation the wrong way round?

apphead, 2018-07-21:
### Version
1.40.1-beta+win64

### Behavior
* _Decreasing XY Size Compensation actually increases hole size_
The comment says otherwise.

zerg960, 2018-07-22:
It makes all the perimeters move inwards, towards the solid infill, if its negative. So a negative number would make the entire object smaller and any holes bigger, to (ideally) compensate for thermal expansion.

rongith, 2018-07-25:
That is a function that I never judge useful. i use only for the first layer (elephant foot compensation). The thermal expansion would be corrected by a ratio and not a fixed number.
@supermerill proposed a unified fixed correction for XY and elephant foot compensations in his fork (https://github.com/supermerill/Slic3r). Then you have to enter -0.2 per example to move the material inward for elephant foot compensation.
It would have been interesting to have and XY compensation for small features with concave only shape like holes as the external perimeter tend to go inward and that is nor the case for convex features.

ufoDziner, 2019-09-18:
Would really like to see this have the capability to have a % value.  Thanks!

bubnikv, 2019-09-18:
> Would really like to see this have the capability to have a % value.

Percent of what? Hole diameter? Slicer currently does not estimate diameters, this would have to be a new functionality.

josedpedroso, 2023-03-30:
> > Would really like to see this have the capability to have a % value.
> 
> Percent of what? Hole diameter? Slicer currently does not estimate diameters, this would have to be a new functionality.

Percentage of total print size on an axis. This would be most useful to compensate for thermal contraction, as explained in [this video](https://www.youtube.com/watch?v=0dFThbwAx2Y), where the printer is already kinematically correct.

closed 2026-09-29 by prusaslicer-bot[bot]