prusa3d/PrusaSlicer #1067: white build plate

Wyldesyde007, 2018-07-21:
### Version
1.40.1

### Operating system type + version
_windows 10 home addition 64bit_

### Behavior
The new build plate graphic shows up as a white plane (if I rotate the view to the underside where it turns to wireframe I can see the graphics)_

[image]
[image]

zerg960, 2018-07-22:
It's not white for me. Might want to check out the files in `C:\Program Files\Prusa3D\Slic3rPE\resources\icons\bed`

Wyldesyde007, 2018-07-22:
Thanks zerg960
I did check out the images in the folder you mentioned and they appear fine it just doesn't show up when I run slic3r.

zerg960, 2018-07-22:
I checked out the code. The path for the textures is hardcoded (in `xs/src/slic3r/GUI/GLCanvas3D.cpp`), and they only get rendered if the selected printer is a Prusa MK2 or MK3. If there's a problem with loading either the top or the bottom texture, or it's not a Prusa, it draws some sort of grid w/o text. Otherwise, depending on the camera elevation, it draws either the top or the bottom texture. What's odd is that your bottom texture works but the top one doesn't, and there's more OpenGL calls for the bottom than there are on the top.

I'm stumped. Could you rule out things like.. installation issues, graphics drivers, and OpenGL stuff? Also, there's a "use Legacy OpenGL 1.1" option in the Slic3r preferences menu that might do something.

Wyldesyde007, 2018-07-22:
I didt do a reinstall this morning as well as switching on the legacy settings which didn't have any effect, I also switched to a diffent model of printer as you just suggested to see if it would appear but that did nothing as well, totally odd.

bubnikv, 2018-07-23:
We have a report from one of college having the same problem on an old laptop she uses at home. She is on vacation, she will be back in two weeks, so we will have a model to debug on.

My bet is, a buggy OpenGL driver is guilty.

vojtechkral, 2018-07-23:
I have the same problem when running Slic3r on Windows in VirtualBox.

bubnikv, 2018-07-23:
Great, we can most likely reproduce it on our Windows 7 box over the remote
desktop. In that case the OpenGL is emulated by the Remote Desktop service
with OpenGL 1.1 compatibility only.

commit 0bd8affa (2018-07-23): Attempt to fix #1067
commit de085fdb (2018-07-23): Power of two squared bed textures to try to fix #1067
closed 2018-07-24 by enricoturri1966