prusa3d/PrusaSlicer #1069: Repository with printer profiles?

petterreinholdtsen, 2018-07-22:
Hi.

Is there a repository somewhere with printer profiles for more 3D printers than the four currently present in /usr/share/slic3r-prusa3d/profiles/?  It would be great if slic3r-prusa had such profiles available, similar to the way Cura provide a selection of profiles for a lot of 3D printers.  If such profile existed for Ultimaker 2+, I would dare to use slic3r-prusa myself. :)

Such collection of profiles would not have to be part of the slic3r-prusa source distribution, it would be maintained elsewhere.

bubnikv, 2018-07-23:
we do not actively maintain profiles for 3rd party printers, but the Slic3r
PE supports that. You are most welcome to provide a vendor specific
configuration bundle.

You may find inspiration in our own profiles:
https://github.com/prusa3d/Slic3r/tree/master/resources/profiles

vojtechkral, 2018-07-23:
Related issue (that also has documentation): #934

petterreinholdtsen, 2018-07-23:
OK, thanks for the quick response.  I lack the skill set and required spare time to make such profile myself, so I will have to stick with Cura for now.

bubnikv, 2018-07-23:
ok, closing this request, leaving #934 open.

closed 2018-07-23 by bubnikv