prusa3d/PrusaSlicer #1070: MMU2 purge block line in mid air

mcm001, 2018-07-24:
### Version
_Version of Slic3r used goes here_
1.41.0 Alpha 2
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
### Behavior
Load two or more objects and set them to different extruders, one taller than the other
Observe prime tower preview gcode
Find the floating line of prime in mid air made of taller part color

[image]

_Is this a new feature request?_
No
#### STL/Config (.ZIP) where problem occurs - any two stls with one taller than the other, am not permitted to upload pictured STLs.

_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Config is stock mk3 mmu config (no changes made)

lukasmatena, 2018-07-24:
As far as I can tell, there is nothing wrong. What you see is the final unload of the filament, so the last filament does not solidify in the nozzle after the print terminates.

mcm001, 2018-07-24:
sounds reasonable to me. I was just confused when I saw extrusion in mid air. This issue can be closed.

closed 2018-07-24 by mcm001

bubnikv, 2018-07-24:
great. Good luck with the MM.