prusa3d/PrusaSlicer #1072: Extra commands in Vase mode: seam on vase as result

alsopub, 2018-07-24:
### Version
Slic3r PE 1.41.0-alpha2, 1.40.1

### Operating system type + version
MacOS 10.9.5

### Behavior
I see a "little seam" on my vases.
I think it is because there are some extra commands in G-Code.

Here is a fragment of "spiral" section of G-Code:

> G1 Z38.000 X160.042 Y65.810 E5218.63620
> G1 Z38.000 X160.202 Y66.317 E5218.66968
> G1 Z38.000 X160.309 Y66.767 E5218.69877
> G1 Z38.000 F7800.000
> ; PURGING FINISHED
> G1 F2177.16
> G1 Z38.000 X160.452 Y67.329 E5218.73540
> G1 Z38.000 X160.511 Y67.866 E5218.76936
> G1 Z38.001 X160.532 Y68.295 E5218.79641 

Look like this command (repeated on each 0.2mm (layer height)):
> G1 Z38.000 F7800.000

actually does not do anything, but printer anyway execute it and make a little delay and as result - a little seam on vase (not actually seam, but visible defect - some extra plastic leaked from nozzle at this time).

Not sure if:
> G1 F2177.16

also affect on delay, but I think it also may be removed or integrated into next G1 command.

#### STL/Config (.ZIP) where problem occurs
Any closed object, "Spiral vase" printing mode.

bubnikv, 2018-07-25:
Well, you are right that there is this unnecessary Z up movement, but it does nothing, so a good firmware should just ignore it. What firmware are you running on?

alsopub, 2018-07-25:
My printer is Anet A6 (V 2.0, stock).
I'm not sure what firmware Anet installed on it.
Maybe the problem is in slow 8bit CPU.

bubnikv, 2018-08-02:
What you may see is actually not a problem of the tiny delay, but the way how Slic3r (and other slicers) implement the vase mode in a hackish way. The slicers usually slice an object with parallel layers, and then they artificially connect end of one perimeter to the start of the perimeter on the successive layer. This will naturally lead to an abrupt step, which depends on the vertical angle.

Is this explanation understandable?

alsopub, 2018-08-03:
There is no Z position change near X160 Y66.
I will fix this exproblem for myself with script (with "PURGING FINISHED" and without).
Then I will print the same model in both ways and show the result.

alsopub, 2018-08-03:
I printed [this vase](https://www.thingiverse.com/thing:687931), and [here is "seam" photo](http://joxi.ru/KAx5Ew3UM7kpqr.png).
It is too difficult to take a good photo, sorry for quality.
After removing extra Z moving commands, seam eliminated.
I'm not require you to fix this, because I can fix this by myself.
If you consider this is NOT a problem - just close this issue.

closed 2026-09-29 by prusaslicer-bot[bot]