prusa3d/PrusaSlicer #1073: Slic3r not writing proper temperature in code

JorgenJensen, 2018-07-25:
### Version
_Version of Slic3r used goes here_
Slic3r Prusa Edition - 1.40.1+win65
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
I have created my own filament settings for temperature. When I slice my object the code writes the temperature for PLA. 
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-07-26:
Would you please update your AMF or 3mf, and the generated G-code? Thanks.

FidelCapo, 2018-07-26:
I tried to reproduce this issue, but I did not. Can you please send the gcode in which the error is displayed and all steps needs to reproduce this? Thank you.

monotan, 2018-08-26:
I wanted to confirm I am seeing similar issue with filament temperature not being correctly written to the gcode file.

Slic3rPE v1.41.0-beta+ / MacOS.

- Slice model
- Export to GCode
- using OctoPrint or SD card to get the file to the printer
- wrong filament temperature is set at the printer
- inspecting GCode file show lower temperature
- however, bed temperature is correctly set
- work around is to manually set filament temperature once the print is started which is a PIA

Let me know if I can provide any more details to help resolve this issue.

FidelCapo, 2018-08-30:
@monotan Unfortunately, I'm still not able to reproduce this issue. Can you please send me config.ini of settings at which this error occurs ? Thank you

monotan, 2018-08-30:
Please find the attached files: config.ini, screenshot of the Filament Temperature screen, and the produced gcode file. Please let me know if you need anything else.

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/2337090/config.ini.zip)
[image]
[3DBenchy.gcode.zip](https://github.com/prusa3d/Slic3r/files/2337092/3DBenchy.gcode.zip)

Braccoz, 2018-09-01:
i have this issue too.

basically, the "other layer" temperature is completely ignored, and the "first layer" temperature is set in the gcode, BUT -5°. (so if i set 205 it sets 200, if i set 203 it sets 198 and so on).

still present on 1.41.0

Braccoz, 2018-09-02:
found the culprit: i had "ooze prevention" on. So basically the ooze prevention option is bugged, and does not produce appropriate gcode.

FidelCapo, 2018-09-10:
If I understand everything correctly, there are two issues here.
The first issue did not change the temperature when the material was changed. This is fixed and fix will be part of the next release.
The second “issue” concerns the ooze prevention function that I think works properly. If this feature is turned on and a single extruder or multimaterial printer with one extruder is set, the function will not be reflected.

Braccoz, 2018-09-10:
not correct. i have 2 extruders on my printer. say i slice a print that only uses 1 extruder. then the bug appears, when in reality the setting should be ignored

FidelCapo, 2018-09-10:
@Braccoz Can you send me a configuration file? Thank you

Flav1u, 2018-10-27:
Post again from the prusa mmu2 forum ...

Ok know I am pretty sure there is some sort of mixup with temperature changes with the mmu2 unit. And it is reproducible.

I've set up:
PLA: 1. Layer = 215°; X. Layer 205°
BVOH: 1. Layer 200°; X. Layer 200°

First time it prints BVOH it does it with 205°.
After that it prints both with the correct temperature.

But there seems to be a problem with the ramming sequence. It does switch to the temperature for the next filament before it starts ramming for the actual filament. So it's always ramming with the wrong temperature.

Files:
[files.zip](https://github.com/prusa3d/Slic3r/files/2522183/files.zip)

lukasmatena, 2018-10-27:
You are correct that the first time it prints BVOH, it uses 205 degress, even though it should not. I'll have a look at it when I have time.

However, you are NOT correct that ramming is done at the wrong temperature - the M104 command is always issued just after the ramming, as explained in https://github.com/prusa3d/Slic3r/issues/1339.

JorgenJensen, 2018-10-29:
Thank you. I actually figured out what was happening.  In the main screen,
you have to make sure that you have selected the filament you want.

bubnikv, 2019-01-28:
@JorgenJensen Could we close this issue then?

JorgenJensen, 2019-01-31:
Yes

bubnikv, 2019-01-31:
Cool, closing.

closed 2019-01-31 by bubnikv