prusa3d/PrusaSlicer #1074: Slic3r PE 1.41.0-alpha2 crashes with certain STL in Windows (not MacOS)

kazibole, 2018-07-25:
### Version
Slic3r 1.41.0-alpha2

### Operating system type + version
Windows 10 (problem)
MacOS 10.13 (no problem)

### Behavior
* _Describe the problem_
Attempt to open the STLs linked below, Slic3r on Windows 10 will crash before fully opening the STL. Slic3r on MacOS 10.13 opens the file without issue. Opening the STL in Meshmixer and exporting it to a new STL file allows Slic3r on Windows 10 to open it.

* _Expected Results_
The STL is opened successfully. 

* _Actual Results_
Slic3r crashes before fully opening the STL.

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
https://www.thingiverse.com/thing:2446648
https://www.thingiverse.com/thing:2050885

FidelCapo, 2018-07-30:
Thank you for the report. I think this problem is caused by the Arrangement function. If the auto-center parts is enabled in the Slic3rPE preferences, this error occurs immediately when the object is imported. Can you please try import this object when the automatic arrangement is disabled?

Thank you.

kazibole, 2018-07-30:
@FidelCapo I tried it with auto centering off, and it imported without a problem.

kazibole, 2018-08-07:
@FidelCapo @bubnikv Confirmed resolves in 1.41.0-beta!

closed 2018-08-07 by kazibole