prusa3d/PrusaSlicer #1076: Broken sliced model, 3D view is ok  - Version 1.41.0-alpha2+linux64 and also in MacOS

vernieg, 2018-07-26:
### Version
1.41.0-alpha2+linux64
1.41.0-alpha2 MacOS

### Operating system type + version
Linux mbox 4.18.0-1-MANJARO #1 SMP PREEMPT Mon Jul 16 06:26:07 UTC 2018 x86_64 GNU/Linux
MacOS high Sierra

### Behavior
Broken tip of the model under linux and MacOS for the same Version 1.41.0-alpha2.

* _Steps needed to reproduce the problem_
Orient the model 180 degrees rotation on X axis.
Just slice the model from https://www.thingiverse.com/thing:2932168/files or the attached STL file

* _Expected Results_
There should be no broken material on the tip of the model.
Note: It did sliced well with Cura.

* _Actual Results_
Broken material on the tip of the model.
3D view:
[image]
Sliced preview:
[image]

#### STL/Config (.ZIP) where problem occurs
[stl-and-config.zip](https://github.com/prusa3d/Slic3r/files/2232451/stl-and-config.zip)

bubnikv, 2018-08-02:
Will it slice correctly with the "detect thin walls" feature enabled?

vernieg, 2018-08-03:
The "detect thin walls" feature is enabled but did not slice correctly even on the current alpha3 version.

vernieg, 2018-08-03:
Ok, if I disabled the "detect thin walls", it did sliced correctly. :) 
[image]

bubnikv, 2018-08-03:
I think we can close the issue then?

vernieg, 2018-08-03:
Yes, I think this can be considered closed. I just have to pay attention to these advance settings which is fine with me. :) Thank you.

bubnikv, 2018-08-03:
Thanks. Closing.

closed 2018-08-03 by bubnikv