prusa3d/PrusaSlicer #1077: Feature request: Right click options for objects in list

Boby71, 2018-07-26:
### Version
1.41.0-alpha2+win64

### Operating system type + version
Win10/64 Bit

### Behavior
This feature request is described easily:
In the object list on the right side it should be possible to add a context-sensitive menu when a right mouse click has been performed.

[image]

It should offer at least:
- Delete object

It would be nice to have a "lock/unlock" object function, to exclude one or more objects from the "delete all" function (or any other functions who are being applied to all objects on the platter).

Other useful right-click actions might make sense, but the missing delete function is currently a pain, when you have a STL file with many seperate parts - and you only need one of them. I just had some parts being very small and not easily visible spread everywhere (also outside the print area) - and to find and click them on the platter view was really annoying.

bubnikv, 2018-08-02:
@YuSanka is working on this part of UI as of now. We plan to have the "Object settings" window functionality moved into that object list in a tree form.

Boby71, 2018-08-02:
That‘s amazing news, thank you!

LG,
Robert

bubnikv, 2018-12-12:
The soon to be released 1.42 alpha supports
- delete key (or the backspace on the low cost products not offering a delete key) in both the 3D scene and the object list
- context menu on both the 3D scene and the object list

We did not implement the "hiding" of the objects from the slicing process by an object attribute, but now the objects fully outside of the plater are ignored by the slicing engine. Does this solution fulfill your needs?

Boby71, 2018-12-25:
From my side, this ticket can be closed - 1.42.0 offers these features already, thanks for the great work!

closed 2018-12-25 by Boby71