prusa3d/PrusaSlicer #1078: Slic3r 1.41.0-alpha2+18 - M73 time estimator issue.

rtyr, 2018-07-26:
### Version
1.41.0-alpha2+18

### Operating system type + version
Windows 10 64bit

### Behavior
MMU, 3 models, wipe to infill/wipe to object enabled.
Slic3r time estimation = 11:36

Remaining time on the LCD was 00:38R at 0%. At roughly 5% (Z1.2) the value has changed to +/- 11:00R.
[image]

#### STL/Config (.ZIP) where problem occurs
AMF/G-CODE:
[SnailBug.zip](https://github.com/prusa3d/Slic3r/files/2233346/SnailBug.zip)

bubnikv, 2018-07-27:
@PavelSindler It looks like a firmware bug. It looks like the first value
shown has not been correctly initialized?

PavelSindler, 2018-07-27:
I will check it, thank you for feedback.

PavelSindler, 2018-07-27:
Which firmware version do you have @rtyr ?

bubnikv, 2018-08-02:
I am closing the issue here at Slic3r, the issue seems to be on the firmware side.

closed 2018-08-02 by bubnikv