prusa3d/PrusaSlicer #1082: Feature request: Rotate X Button 90deg

KnatteAnka, 2018-07-27:
Button or key command to rotate object 90deg in X rotation becase that is some thing that i do for more the 25% of my prints! insted of dropdown and write 90 or 180 deg.

dikodahan, 2018-07-27:
Definitely a +1 for this one!!!

BirkBinnard, 2018-07-27:
[image]

bubnikv, 2018-08-02:
We are working on a PrusaControl like UI solution
https://prusacontrol.org/
with a contextual transformation panel at the right.

bubnikv, 2018-12-12:
We have implemented 3D gizmos in Slic3r 1.42 alpha, which is to be released soon. We have also implemented a transformation spreadsheet similar to the prusa control. I believe these two changes should fit your bill. Closing.

closed 2018-12-12 by bubnikv

BirkBinnard, 2018-12-12:
Excellent news - thanks for the info. Looking forward to your new release.