prusa3d/PrusaSlicer #1083: [1.41.0-alpha2] Window panes changing when switching between first or last listed printer

fiveangle, 2018-07-27:
### Version
1.41.0-alpha2 (https://github.com/prusa3d/Slic3r/commit/68cd51435f2e5b830156534a144efe64bc019b7d)

### Operating system type + version
MacOS 10.13.4

### Behavior

After install and creation of new configurations dialog is accepted, the first printer in the list is selected.  Selecting different printers resizes the window panes without reason.  I have not determined the algorithm but it seems to have to do with selecting the last printer entry toggling some internal variable that then toggles the window pane sizes when other printer is later selected (and back and forth).

### Steps to recreate
- Launch for the first time
- Allow creation of new config profiles [should show only 4 default profiles, MK2, MK2-0.25, MK3, MK3-0.25]
- Change _printer_ type randomly
- Observe window panes resize pseudorandomly

Screencap:
https://www.screencast.com/t/mwxImt5WsX

bubnikv, 2018-08-02:
@YuSanka @FidelCapo would you please look into this issue? Thanks.

bubnikv, 2018-09-01:
@YuSanka we should make the width constant, independent of the content of the column.