prusa3d/PrusaSlicer #1084: Clicking Arrange causes crash

BirkBinnard, 2018-07-27:
### Version
1.41.0-alpha2

### Operating system type + version
Windows 10

### Behavior
After loading any STL file, clicking the Arrange button  crashes the program. Boom! - it just totally disappears. 

I got this info from Event Viewer's Application log:
Faulting application name: slic3r.exe, version: 1.41.0.0, time stamp: 0x5b535eb3
Faulting module name: XS.dll, version: 0.0.0.0, time stamp: 0x5b535e1d
Exception code: 0xc0000005
Fault offset: 0x0000000000283b87
Faulting process id: 0x2964
Faulting application start time: 0x01d425f4f6f69e58
Faulting application path: C:\Program Files\Plic3r\slic3r.exe
Faulting module path: C:\PROGRA~1\Plic3r\lib\auto\Slic3r\XS\XS.dll
Report Id: 7f90819c-0376-4e2d-97bb-4c53b323d2fd
Faulting package full name: 
Faulting package-relative application ID: 

#### STL/Config (.ZIP) where problem occurs
I got the same behavior from a number of different STL files. Big & small, simple & complex - it didn't seem to make a difference.

DrLex0, 2018-07-28:
Same here in OS X 10.11.6 with 1.41.0-alpha3
[Crash.zip](https://github.com/prusa3d/Slic3r/files/2238014/Crash.zip)

Stoebei, 2018-07-29:
I have the same Problem. Are you using a round Plate?

BirkBinnard, 2018-07-29:
If you are asking me the answer is yes - I have a delta style printer with a round print bed.

DrLex0, 2018-07-29:
I have a rectangular print bed. The 0, 0 coordinate is at the center.

Stoebei, 2018-07-29:
I have 2 Profiles. The Square one got the Origin on the left front Corner of the Bed, this works.                        The Delta one crashes when i press the arrange Button or enable the auto Arrange option.