prusa3d/PrusaSlicer #1085: Crash on importing big stl

OBELIKS, 2018-07-28:
### Version
1.41.0-alpha3+win64

### Operating system type + version
Windows 10 64bit

### Behavior
Slic3r crashes when importing too big stl, amf or obj. 3mf has no problems.

_Is this a new feature request?_ No

#### STL/Config (.ZIP) where problem occurs
[slic3r_test.zip](https://github.com/prusa3d/Slic3r/files/2237838/slic3r_test.zip)

kazibole, 2018-07-28:
Opens fine in 1.41.0-alpha3 on MacOS 10.13. Sounds like the same issue as in #1074.

BirkBinnard, 2018-07-28:
I loaded all 4 versions of the test file with no problem on my Win10 16 GB RAM system. (What is an amf file anyway?) And just to make sure I loaded an old file of mine that is 109 MB in size. No problem with that one either.

FidelCapo, 2018-07-30:
Thank you for the report. I think this problem is caused by the Arrangement function. If the auto-center parts is enabled in the Slic3rPE preferences, this error occurs immediately when the object is imported. Can you please try import this object when the automatic arrangement is disabled?

Thank you.

OBELIKS, 2018-07-30:
Confirmed. when I turn Auto-center parts off, models load.

FidelCapo, 2018-07-30:
@OBELIKS Thank you