prusa3d/PrusaSlicer #1086: Q: acceleration issue with non prusa printer

boelle, 2018-07-28:
### Version
1.41.0-alpha2+win64

### Operating system type + version
win10

### Behavior
* _Describe the problem_

i used to use version 1.39.2 and was very happy with it

i use it together witn my mendel90 and marlin bugfix 2.0.x with the same feedrate, acceleration and jerk settings as for a mk2s taken from here: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/1_75mm_MK25-RAMBo13a-E3Dv6full.h 
and here: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Configuration.h

then the youtuber cnc kitchen showed the gyroid infill and i found out that i have to update, no big deal i thought, but i was wrong

in version 1.39.2 i pretty much just copied the settings etc for the Mk2s and changed the start gcode, that was it

but in 1.41 in the gcode it adds a section in the start with feedrates etc and then every now and then a progress code

now my printer skips steps (at least during non print moves) because it moves to fast and i just cant figure how to get rid of that section in the start. i filled in the machine limits with just zero but then it moves so slow i cant see it moves

i even tried to create a custom setup from scratch but the same thing happend

any pinters or suggestions are welcome... and yes i know i dont have a prusa printer but it like the software :-/

Stoebei, 2018-07-31:
I think the easiest Way is to type your Firmware Values in the Machine Limits.

Feedrate X+Y 200 Z12 E120
Accel 1000 1000 200 5000
Print and Retract Accel 1250
And Jerk 10 10 0.4 2.5
Then Slic3r use the same Values as your Firmware

boelle, 2018-08-01:
i tried that as the first thing, of course i could fill in 0 for all the machine limits and it then moved so slow i could not see it but only hear it and feel it

bubnikv, 2018-08-02:
I think your issue is the same as https://github.com/prusa3d/Slic3r/issues/1089
The issue is caused by the change in the M204 behavior in the mainline marlin after we forked from it.
Would you please confirm that?

boelle, 2018-08-02:
not sure how to confirm it

boelle, 2018-08-02:
the only difference that was clear to me was the gcode that comes out of the latest Slic3r

latest has a section in the start with different codes that overrides the feedratest etc
the versions before that do not insert that 

i would have used prusa firmware  but on my selfbuild mendel90 i use dual z steppers and dual endstop and i have a full graphic display, and if i rememeber correct prusa version of marlin does not support that

bubnikv, 2018-08-03:
Marlin changed the definition of the M204 command, see the discussion at 
https://github.com/prusa3d/Slic3r/issues/1089

For the time estimator to work, the Slic3r needs to know the machine limit values with which the firmware operates. To keep both in sync, Slic3r now exports the machine limits as M-code block before the custom start G-code. The upcoming release will generate an M-code compatible with the upstream Marlin, so you should be good to go as long as you enter meaningful values into the machine limits table.

I am closing this issue, let's continue at https://github.com/prusa3d/Slic3r/issues/1089 if you find any remaining issue with the upcoming release.

closed 2018-08-03 by bubnikv

boelle, 2018-08-03:
great, will give it a go as soon i see the next version, again thanks for this great software