prusa3d/PrusaSlicer #1087: Crash on Help Menu

Mike4U, 2018-07-28:
1.40.1+
MacOS 10.13.6
Crashes when menu pulldown selected: Help -> System Info

BirkBinnard, 2018-07-28:
Works OK on Win10.

tracernz, 2018-07-28:
Works on 1.41.0-alpha3 with macOS 10.14 Beta (18A336e)
```
Slic3r Prusa Edition
Version: 1.41.0-alpha3+
Build:   Slic3rPE-1.41.0-alpha3+full-201807272036
Operating System:    darwin
System Architecture: darwin-thread-multi-2level
System Version: Darwin macbook.bertie.nz 18.0.0 Darwin Kernel Version 18.0.0: Thu Jul 12 19:03:47 PDT 2018; root:xnu-4903.200.304.41.2~2/RELEASE_X86_64 x86_64
```

This was fixed in https://github.com/prusa3d/Slic3r/issues/1034.. Which GPU do you have?

Mike4U, 2018-07-29:
Sorry, I somehow loaded 1.40, I think from the Prusa download page.  I even edited my first post to change it to 1.41.   It doesn't help that the typical way of finding versions on MacOS is apple-I at the finder level which doesn't give enough info in Slic3r.

Regardless, it works now, and thanks all for a great program:
Slic3r Prusa Edition
Version: 1.41.0-alpha3+
Build:   Slic3rPE-1.41.0-alpha3+full-201807272036
Operating System:    darwin
System Architecture: darwin-thread-multi-2level
System Version: Darwin Michaels-iMac.home 17.7.0 Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64 x86_64

Summary of my perl5 (revision 5 version 26 subversion 1) configuration:
   
  Platform:
    osname=darwin
    osvers=13.4.0
    archname=darwin-thread-multi-2level
    uname='darwin mavericks-build-vm.local 13.4.0 darwin kernel version 13.4.0: sun aug 17 19:50:11 pdt 2014; root:xnu-2422.115.4~1release_x86_64 x86_64 '
    config_args='-de -Dprefix=/Users/prusa3d/perl5/perlbrew/perls/perl-5.26.1 -Duserelocatableinc -Dusethreads -Aeval:scriptdir=/Users/prusa3d/perl5/perlbrew/perls/perl-5.26.1/bin'
    hint=recommended
    useposix=true
    d_sigaction=define
    useithreads=define
    usemultiplicity=define
    use64bitint=define
    use64bitall=define
    uselongdouble=undef
    usemymalloc=n
    default_inc_excludes_dot=define
    bincompat5005=undef
  Compiler:
    cc='cc'
    ccflags ='-fno-common -DPERL_DARWIN -mmacosx-version-min=10.9 -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -I/opt/local/include -DPERL_USE_SAFE_PUTENV'
    optimize='-O3'
    cppflags='-fno-common -DPERL_DARWIN -mmacosx-version-min=10.9 -fno-strict-aliasing -pipe -fstack-protector -I/usr/local/include -I/opt/local/include'
    ccversion=''
    gccversion='4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)'
    gccosandvers=''
    intsize=4
    longsize=8
    ptrsize=8
    doublesize=8
    byteorder=12345678
    doublekind=3
    d_longlong=define
    longlongsize=8
    d_longdbl=define
    longdblsize=16
    longdblkind=3
    ivtype='long'
    ivsize=8
    nvtype='double'
    nvsize=8
    Off_t='off_t'
    lseeksize=8
    alignbytes=8
    prototype=define
  Linker and Libraries:
    ld='cc'
    ldflags =' -mmacosx-version-min=10.9 -fstack-protector -L/usr/local/lib -L/opt/local/lib'
    libpth=/usr/local/lib /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/6.0/lib /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib /usr/lib /opt/local/lib
    libs=-lpthread -ldbm -ldl -lm -lutil -lc
    perllibs=-lpthread -ldl -lm -lutil -lc
    libc=
    so=dylib
    useshrplib=false
    libperl=libperl.a
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_dlopen.xs
    dlext=bundle
[... 135 lines cut ...]
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_multisample_filter_hint
GL_NV_point_sprite
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

closed 2018-07-29 by Mike4U